tianocore/edk2 #10896: [Bug]: UefiPayloadPkg: Image Section Alignment does not match Required Alignment

radiofan, 2025-03-27:
### Is there an existing issue for this?

- [x] I have searched existing issues

### Bug Type

- [x] Firmware
- [ ] Tool
- [ ] Unit Test

### What packages are impacted?

UefiPayloadPkg

### Which targets are impacted by this bug?

DEBUG

### Current Behavior

The load log contains many messages "!!!!!!!!  Image Section Alignment(0x40) does not match Required Alignment (0x1000)  !!!!!!!!". This means that the CreateImagePropertiesRecord() function fails.

https://github.com/tianocore/edk2/blob/9006a9b5e4671ffc3d29c035260b19a8075c3f64/MdeModulePkg/Library/ImagePropertiesRecordLib/ImagePropertiesRecordLib.c#L1078-L1088

[load.log](https://github.com/user-attachments/files/19483476/load.log)

### Expected Behavior

Eliminate errors

### Steps To Reproduce

Build universal (or non universal) payload:
```bash
cd /edk
source edksetup.sh
python UefiPayloadPkg/UniversalPayloadBuild.py -a X64 -b DEBUG -t GCC5
```
In my case, the payload bootloader is SBL. Testing was performed on a hardware.

[default.build.log](https://github.com/user-attachments/files/19483592/default.build.log)

### Build Environment

```markdown
- OS: Linux version 6.8.0-52-generic (buildd@lcy02-amd64-099) (x86_64-linux-gnu-gcc-12 (Ubuntu 12.3.0-1ubuntu1~22.04) 12.3.0, GNU ld (GNU Binutils for Ubuntu) 2.38) #53~22.04.1-Ubuntu SMP PREEMPT_DYNAMIC Wed Jan 15 19:18:46 UTC 2
- Tool Chain: gcc (Ubuntu 13.1.0-8ubuntu1~22.04) 13.1.0
- Tool Chain: Ubuntu clang version 14.0.0-1ubuntu1.1
- Tool Chain: GNU ld (GNU Binutils for Ubuntu) 2.38
```

### Version Information

```text
b279ac9f36e1b386c8c8a761657c6e6d0f1dfa1d
```

### Urgency

Low

### Are you going to fix this?

I will fix it

### Do you need maintainer feedback?

Maintainer feedback requested

radiofan, 2025-03-27:
[https://github.com/tianocore/edk2/pull/10897](https://github.com/tianocore/edk2/pull/10897)

radiofan, 2025-03-27:
Currently, the alignment is set only for DXE_RUNTIME_DRIVER. Next drivers (and applications) also need 4KB alignment. The list was obtained by analyzing the [load log](https://github.com/user-attachments/files/19483476/load.log).

```
FatPkg/EnhancedFatDxe/Fat/DEBUG/Fat.dll
MdeModulePkg/Bus/Ata/AtaAtapiPassThru/AtaAtapiPassThru/DEBUG/AtaAtapiPassThruDxe.dll
MdeModulePkg/Bus/Ata/AtaBusDxe/AtaBusDxe/DEBUG/AtaBusDxe.dll
MdeModulePkg/Bus/Isa/Ps2KeyboardDxe/Ps2KeyboardDxe/DEBUG/Ps2KeyboardDxe.dll
MdeModulePkg/Bus/Isa/Ps2MouseDxe/Ps2MouseDxe/DEBUG/Ps2MouseDxe.dll
MdeModulePkg/Bus/Pci/EhciDxe/EhciDxe/DEBUG/EhciDxe.dll
MdeModulePkg/Bus/Pci/NvmExpressDxe/NvmExpressDxe/DEBUG/NvmExpressDxe.dll
MdeModulePkg/Bus/Pci/PciBusDxe/PciBusDxe/DEBUG/PciBusDxe.dll
MdeModulePkg/Bus/Pci/PciHostBridgeDxe/PciHostBridgeDxe/DEBUG/PciHostBridgeDxe.dll
MdeModulePkg/Bus/Pci/SataControllerDxe/SataControllerDxe/DEBUG/SataController.dll
MdeModulePkg/Bus/Pci/SdMmcPciHcDxe/SdMmcPciHcDxe/DEBUG/SdMmcPciHcDxe.dll
MdeModulePkg/Bus/Pci/UhciDxe/UhciDxe/DEBUG/UhciDxe.dll
MdeModulePkg/Bus/Pci/XhciDxe/XhciDxe/DEBUG/XhciDxe.dll
MdeModulePkg/Bus/Scsi/ScsiBusDxe/ScsiBusDxe/DEBUG/ScsiBus.dll
MdeModulePkg/Bus/Scsi/ScsiDiskDxe/ScsiDiskDxe/DEBUG/ScsiDisk.dll
MdeModulePkg/Bus/Sd/EmmcDxe/EmmcDxe/DEBUG/EmmcDxe.dll
MdeModulePkg/Bus/Sd/SdDxe/SdDxe/DEBUG/SdDxe.dll
MdeModulePkg/Bus/Usb/UsbBusDxe/UsbBusDxe/DEBUG/UsbBusDxe.dll
MdeModulePkg/Bus/Usb/UsbKbDxe/UsbKbDxe/DEBUG/UsbKbDxe.dll
MdeModulePkg/Bus/Usb/UsbMassStorageDxe/UsbMassStorageDxe/DEBUG/UsbMassStorageDxe.dll
MdeModulePkg/Bus/Usb/UsbMouseDxe/UsbMouseDxe/DEBUG/UsbMouseDxe.dll
MdeModulePkg/Core/Dxe/DxeMain/DEBUG/DxeCore.dll
MdeModulePkg/Universal/Acpi/AcpiTableDxe/AcpiTableDxe/DEBUG/AcpiTableDxe.dll
MdeModulePkg/Universal/BdsDxe/BdsDxe/DEBUG/BdsDxe.dll
MdeModulePkg/Universal/Console/ConPlatformDxe/ConPlatformDxe/DEBUG/ConPlatformDxe.dll
MdeModulePkg/Universal/Console/ConSplitterDxe/ConSplitterDxe/DEBUG/ConSplitterDxe.dll
MdeModulePkg/Universal/Console/GraphicsConsoleDxe/GraphicsConsoleDxe/DEBUG/GraphicsConsoleDxe.dll
MdeModulePkg/Universal/Console/TerminalDxe/TerminalDxe/DEBUG/TerminalDxe.dll
MdeModulePkg/Universal/DevicePathDxe/DevicePathDxe/DEBUG/DevicePathDxe.dll
MdeModulePkg/Universal/Disk/DiskIoDxe/DiskIoDxe/DEBUG/DiskIoDxe.dll
MdeModulePkg/Universal/Disk/PartitionDxe/PartitionDxe/DEBUG/PartitionDxe.dll
MdeModulePkg/Universal/Disk/UnicodeCollation/EnglishDxe/EnglishDxe/DEBUG/EnglishDxe.dll
MdeModulePkg/Universal/DisplayEngineDxe/DisplayEngineDxe/DEBUG/DisplayEngine.dll
MdeModulePkg/Universal/EbcDxe/EbcDxe/DEBUG/EbcDxe.dll
MdeModulePkg/Universal/HiiDatabaseDxe/HiiDatabaseDxe/DEBUG/HiiDatabase.dll
MdeModulePkg/Universal/MemoryTest/NullMemoryTestDxe/NullMemoryTestDxe/DEBUG/NullMemoryTestDxe.dll
MdeModulePkg/Universal/Metronome/Metronome/DEBUG/Metronome.dll
MdeModulePkg/Universal/PCD/Dxe/Pcd/DEBUG/PcdDxe.dll
MdeModulePkg/Universal/PlatformDriOverrideDxe/PlatformDriOverrideDxe/DEBUG/PlatDriOverrideDxe.dll
MdeModulePkg/Universal/SecurityStubDxe/SecurityStubDxe/DEBUG/SecurityStubDxe.dll
MdeModulePkg/Universal/SerialDxe/SerialDxe/DEBUG/SerialDxe.dll
MdeModulePkg/Universal/SetupBrowserDxe/SetupBrowserDxe/DEBUG/SetupBrowser.dll
MdeModulePkg/Universal/SmbiosDxe/SmbiosDxe/DEBUG/SmbiosDxe.dll
MdeModulePkg/Universal/WatchdogTimerDxe/WatchdogTimer/DEBUG/WatchdogTimer.dll
OvmfPkg/SioBusDxe/SioBusDxe/DEBUG/SioBusDxe.dll
PcAtChipsetPkg/HpetTimerDxe/HpetTimerDxe/DEBUG/HpetTimerDxe.dll
ShellPkg/Application/Shell/Shell/DEBUG/Shell.dll
UefiCpuPkg/CpuDxe/CpuDxe/DEBUG/CpuDxe.dll
UefiCpuPkg/CpuIo2Dxe/CpuIo2Dxe/DEBUG/CpuIo2Dxe.dll
UefiPayloadPkg/BlSupportDxe/BlSupportDxe/DEBUG/BlSupportDxe.dll
UefiPayloadPkg/GraphicsOutputDxe/GraphicsOutputDxe/DEBUG/GraphicsOutputDxe.dll
```

[This fix](https://github.com/tianocore/edk2/pull/10897) adds alignment for the following modules.
```
FatPkg/EnhancedFatDxe/Fat/DEBUG/Fat.dll
MdeModulePkg/Bus/Ata/AtaAtapiPassThru/AtaAtapiPassThru/DEBUG/AtaAtapiPassThruDxe.dll
MdeModulePkg/Bus/Ata/AtaBusDxe/AtaBusDxe/DEBUG/AtaBusDxe.dll
MdeModulePkg/Bus/Isa/Ps2KeyboardDxe/Ps2KeyboardDxe/DEBUG/Ps2KeyboardDxe.dll
MdeModulePkg/Bus/Isa/Ps2MouseDxe/Ps2MouseDxe/DEBUG/Ps2MouseDxe.dll
MdeModulePkg/Bus/Pci/EhciDxe/EhciDxe/DEBUG/EhciDxe.dll
MdeModulePkg/Bus/Pci/NvmExpressDxe/NvmExpressDxe/DEBUG/NvmExpressDxe.dll
MdeModulePkg/Bus/Pci/PciBusDxe/PciBusDxe/DEBUG/PciBusDxe.dll
MdeModulePkg/Bus/Pci/PciHostBridgeDxe/PciHostBridgeDxe/DEBUG/PciHostBridgeDxe.dll
MdeModulePkg/Bus/Pci/SataControllerDxe/SataControllerDxe/DEBUG/SataController.dll
MdeModulePkg/Bus/Pci/SdMmcPciHcDxe/SdMmcPciHcDxe/DEBUG/SdMmcPciHcDxe.dll
MdeModulePkg/Bus/Pci/UhciDxe/UhciDxe/DEBUG/UhciDxe.dll
[... 26 lines cut ...]
MdeModulePkg/Universal/Metronome/Metronome/DEBUG/Metronome.dll
MdeModulePkg/Universal/MonotonicCounterRuntimeDxe/MonotonicCounterRuntimeDxe/DEBUG/MonotonicCounterRuntimeDxe.dll
MdeModulePkg/Universal/PCD/Dxe/Pcd/DEBUG/PcdDxe.dll
MdeModulePkg/Universal/PlatformDriOverrideDxe/PlatformDriOverrideDxe/DEBUG/PlatDriOverrideDxe.dll
MdeModulePkg/Universal/ReportStatusCodeRouter/RuntimeDxe/ReportStatusCodeRouterRuntimeDxe/DEBUG/ReportStatusCodeRouterRuntimeDxe.dll
MdeModulePkg/Universal/ResetSystemRuntimeDxe/ResetSystemRuntimeDxe/DEBUG/ResetSystemRuntimeDxe.dll
MdeModulePkg/Universal/SecurityStubDxe/SecurityStubDxe/DEBUG/SecurityStubDxe.dll
MdeModulePkg/Universal/SerialDxe/SerialDxe/DEBUG/SerialDxe.dll
MdeModulePkg/Universal/SetupBrowserDxe/SetupBrowserDxe/DEBUG/SetupBrowser.dll
MdeModulePkg/Universal/SmbiosDxe/SmbiosDxe/DEBUG/SmbiosDxe.dll
[... cut ...]
PcAtChipsetPkg/PcatRealTimeClockRuntimeDxe/PcatRealTimeClockRuntimeDxe/DEBUG/PcRtc.dll
ShellPkg/Application/Shell/Shell/DEBUG/Shell.dll
ShellPkg/DynamicCommand/TftpDynamicCommand/TftpDynamicCommand/DEBUG/tftpDynamicCommand.dll
UefiCpuPkg/CpuDxe/CpuDxe/DEBUG/CpuDxe.dll
UefiCpuPkg/CpuIo2Dxe/CpuIo2Dxe/DEBUG/CpuIo2Dxe.dll
UefiPayloadPkg/BlSupportDxe/BlSupportDxe/DEBUG/BlSupportDxe.dll
UefiPayloadPkg/GraphicsOutputDxe/GraphicsOutputDxe/DEBUG/GraphicsOutputDxe.dll
```

[build_with_patch.log](https://github.com/user-attachments/files/19483996/build_with_patch.log)

lgao4, 2025-04-10:
this fix has been approved.

radiofan, 2025-04-14:
> this fix has been approved.

but not merged...