tianocore/edk2 #10901: [Feature]: Improve secure boot error message

MichaelMcDonnell, 2025-03-28:
### Feature Overview

I was trying to boot NixOS on Fedora 41 with the RPM version of Gnome Boxes.  I think it uses this firmware (edk2-ovmf). It failed because secure boot is enforced. It is not enforced in the Flatpak version. I got stuck on it for a couple of days because I couldn't figure out what was going on. I thought there was something wrong with the ISO image I was booting. I've noticed a lot of other people online have had the same issue. The problem is the error message doesn't say that the secure boot signature verification failed.

The last error message when secure boot fails is:
"No bootable option or device was found."

[image]

### Solution Overview

I would like something that indicates that secure boot signature verification failed. Either as the last message or somewhere in the error message chain. For example:
"Secure boot signature verification failed."

### Alternatives Considered

The exact text is not super important as long as it indicates that the secure boot failed.

### What packages are impacted?

OvmfPkg

### Urgency

Low

### Are you going to implement the feature request?

Someone else needs to implement the feature

### Do you need maintainer feedback?

No maintainer feedback needed

### Anything else?

I ended up reporting the issue to NixOS even though it wasn't a NixOS issue https://github.com/NixOS/nixpkgs/issues/393759

lgao4, 2025-04-10:
@ardbiesheuvel , @jyao1 , @kraxel , could you give suggestion for this problem?

closed 2025-06-10 by mergify[bot]

MichaelMcDonnell, 2025-06-10:
Thank you for implementing this! That's great!