tianocore/edk2 #10917: [Feature]: Upgrade OpenSSL to 3.5+ for PQC support

jyao1, 2025-04-03:
### Feature Overview

See UEFI PQC discussion at https://github.com/tianocore/edk2/issues/10279.

[OpenSSL 3.5 coming on April 8](https://openssl-library.org/post/2025-02-04-release-announcement-3.5/) will introduce ML-KEM, ML-DSA and SLH-DSA.

### Solution Overview

Upgrade current current OpenSSL 3.4.1 to 3.5+

### What packages are impacted?

CryptoPkg

### Urgency

Low

### Are you going to implement the feature request?

I will implement the feature

### Do you need maintainer feedback?

No maintainer feedback needed

lgao4, 2025-04-10:
Jiewen will take it.

kraxel, 2025-07-17:
Note that commit https://github.com/tianocore/edk2/pull/10946/commits/cd04f3b230177a28f5e9159142248b86b0b0c21c in that PR turns off pqc support to reduce openssl size.

Can be reverted of course.  Maybe it makes sense to only enable pqc for the `OpensslLibFull*.inf` config variants.

closed 2025-07-17 by mergify[bot]