tianocore/edk2 #10921: CryptoPkg: Remove .CRT section in the nasm file as it is not used.

kanagavels97, 2025-04-04:
Hi @mdkinney @lgao4 

We are facing a .CRT section warning mentioned below during the build of OpensslLibFullAccel library.

[image]

I discovered that the .CRT sections are defined in the x64cpuid.nasm and x86_64cpuid.nasm files. 

x86_64cpuid.nasm

[image]

x86cpuid.nasm

[image]

After commenting them out and rebuilding the source, the build warning was resolved. Seems, this global data is not being used by anyone from the mentioned NASM files.

I came across a comment indicating that this is a harmless warning and can be ignored.
https://github.com/tianocore/edk2/issues/9497#issuecomment-2556217193

We are addressing all warnings and aim to eliminate any build warnings from our build log.

Could you assist in generically removing the .CRT section from these NASM files?

Thanks,
Kanagavel

CC: srinivasanm@ami.com gayathrit@ami.com prarthanasagar@ami.com

mdkinney, 2025-04-04:
They are part of the generated filed from OpenSSL that we do not modify.  If OpenSSL project changes the generated code, we want to use that as is.  

Perhaps the linker scripts could be updated to ignore the .CRT sections?

mdkinney, 2025-04-04:
Perhaps add .crt to the discard list here:

https://github.com/tianocore/edk2/blob/59805c7697c801be8b08e5169bc2300d821c419d/BaseTools/Scripts/GccBase.lds#L87

kanagavels97, 2025-04-21:
Hi @mdkinney 

I added the .CRT file to the discarded section in edk2/BaseTools/Scripts/GccBase.lds and tried again, but the warning persists. This issue is occurring only with the X64 MSFT build.

Could you please advise if commenting out the .CRT section cause any issues?

Thanks,
Kanagavel

kanagavels97, 2025-04-21:
Hi @mdkinney,

Please find below a warning snippet for reference.

[image]

We are facing this warning with the edk2-stable202502 tag as well.

What will be the generic solution to resolve this warning without commenting out the .CRT sections, even though it is not being used?

Please let me know if you need any other information.

Thanks,
Kanagavel

kanagavels97, 2025-06-03:
Hi @mdkinney 

Could you please provide your suggestion for the previous comment?

Thanks,
Kanagavel

mdkinney, 2025-06-10:
I think that syntax is only reserving 16 bytes in .bss section.  I do not see any db/dw/dd statements following that statement to initialize the 16 bytes of .bss content with a value.

It looks like NASM thinks it is being initialized and is moving the symbol to another section.  What happens if you ignore this specific NASM warning for this OpensslLibFullAccel build?

kanagavels97, 2025-06-12:
@mdkinney 

We are using the -w-other flag to suppress the warning for OpenSSLLibFullAccel and have not faced any issues.

kanagavels97, 2025-06-13:
Hi @mdkinney 

Yes, we want this warning to be suppressed, and we will create a PR. I used **MSFT:*_*_*_NASM_FLAGS = -w-other**  
flag in the OpensslLibFullAccel.inf file to suppress the warning. This warning is from the NASM file, and OpensslLibFullAccel and OpensslLibAccel libraries only have support for NASM.

[image]

Could you please help confirm whether the directive MSFT:*_*_*_NASM_FLAGS = -w-other is valid for suppressing .bss section warnings during the build?

Could you please let us know which libraries need to be included to implement the changes?

Thanks,
Kanagavel

kanagavels97, 2025-07-08:
@mdkinney 

Do you have any update on the above comment?