tianocore/edk2 #10922: [Bug]: edksetup.sh doesn't handle spaces in directory name

mythemeria, 2025-04-04:
### Is there an existing issue for this?

- [x] I have searched existing issues

### Bug Type

- [ ] Firmware
- [x] Tool
- [ ] Unit Test

### What packages are impacted?

BaseTools

### Current Behavior

I have cloned the repo into a folder called "not my repo" but the edksetup.sh script fails because there are unquoted directory names

### Expected Behavior

Should not fail when directory names contain spaces

### Steps To Reproduce

Run through setup steps with the repo cloned in a folder that has spaces in the name

### Build Environment

```markdown
- OS(s): Endeavour OS Linux
```

### Version Information

```text
Commit: 59805c7
```

### Urgency

Low

### Are you going to fix this?

Someone else needs to fix it

### Do you need maintainer feedback?

No maintainer feedback needed

### Anything else?

First time using the tool so not confident to fix it properly as I don't know where else this issue may be present, sorry. Obviously can work around it by changing the directory name

lgao4, 2025-04-10:
@mythemeria , this is not bug in edk2 build. I suggest to rename the directory name without space. If you want to support it, it should be new feature.

mythemeria, 2025-04-11:
From a user perspective, this behaves like a bug: the script fails when given a valid and common directory name. Most modern build systems either escape or quote paths to handle this scenario gracefully. I understand it's not a massive issue given there's an workaround, hence why I marked it as low priority, but I don't really agree it's not a bug.

mdkinney, 2025-04-11:
@leiflindholm, @ardbiesheuvel Can you please comment on this issue.  It is not clear to me how many assumptions about no spaces in Linux paths are made in TianoCore and downstream consumers.

ardbiesheuvel, 2025-04-11:
I don't think this ever worked so it is not a regression. I don't feel very strongly about whether we call it a bug or not, given that it has never come up in the 10+ years that I have collaborated on this code base, and so the impact is clearly very low.

Given how thinly we are spread, I don't think we should do anything about this, unless the reporter is willing to contribute a fix themselves.

mythemeria, 2025-04-12:
Like I said I am not super familiar with the tool so I'm not confident to not break something. But maybe at some point in the future when I've used it more I'll give it a shot. Or maybe it could be a good first issue for someone? Anyway no pressure, I understand there's more important stuff on your radar

bexcran, 2025-06-15:
While most *nix users don't use spaces in directory or file names, scripts should ideally be able to handle them.
ShellCheck does warn about adding quotes around variables to handle this, and I'd like to see edksetup.sh and related scripts be free of ShellCheck errors and warnings as far as is possible.

The error you currently get when trying to use edksetup.sh from a directory with spaces in it is:
```
bcran@adder:~/src/tmp/not my repo/edk2$ . ./edksetup.sh 
bash: /home/bcran/src/tmp/not: No such file or directory
```