tianocore/edk2 #10925: [Bug]: Raid card option rom is not displayed in Device Manager form after repairing in Driver Health form.

EricGao2015, 2025-04-07:
### Is there an existing issue for this?

- [x] I have searched existing issues

### Bug Type

- [x] Firmware
- [ ] Tool
- [ ] Unit Test

### What packages are impacted?

MdeModulePkg

### Which targets are impacted by this bug?

DEBUG, RELEASE

### Current Behavior

Raid card option rom is not displayed in Device Manager form after configurating/repairing in Driver Health form.

### Expected Behavior

Raid card option rom is displayed in Device Manager form after repairing in Driver Health form.

### Steps To Reproduce

1. A raid card with option rom is on motherboard. It will install HII resources with `Driver Health` and `Platform Setup`.
2. when raid card is in abnormal state, `Driver Health` is installed but `Platform Setup` is not installed. Enter `Driver Health` form and repair it to make card in normal state.
3. Exit` Driver Health `form, check if raid card option rom is in `Device Manager` form.

### Build Environment

```markdown
- OS(s): CentOS Linux 8
- Tool Chain(s): https://github.com/loongson/build-tools/releases/download/2024.11.01/x86_64-cross-tools-loongarch64-binutils_2.43.1-gcc_14.2.0-glibc_2.40.tar.xz
```

### Version Information

```text
Tag: edk2-stable202411
```

### Urgency

Low

### Are you going to fix this?

I will fix it

### Do you need maintainer feedback?

No maintainer feedback needed

lgao4, 2025-04-10:
@EricGao2015 ,  I will review this change.

closed 2025-04-25 by EricGao2015

mike41xSD, 2026-01-20:
The commit 135e07c breaks UiApp.efi which is built against DeviceManagerUiLib. 
Moving updating  CreateDeviceManagerForm() of the form to the callback from DeviceManagerUiLibConstructor() breaks initialization of the forms after start-up. Entering from "Device Manager"-> to almost any entries of the "Devices List" is impossible, sometimes leading to an exception.

EricGao2015, 2026-01-21:
> The commit [135e07c](https://github.com/tianocore/edk2/commit/135e07c6e514a24a06e74db79a173c72c8d15019) breaks UiApp.efi which is built against DeviceManagerUiLib. Moving updating [...]

Hi Mike, I don't reproduce the issue you mentioned. Can you provide more information how to reproduce it?

EricGao2015, 2026-01-22:
> The commit [135e07c](https://github.com/tianocore/edk2/commit/135e07c6e514a24a06e74db79a173c72c8d15019) breaks UiApp.efi which is built against DeviceManagerUiLib. Moving updating [...]

> In my opinion, missing of **CreateDeviceManagerForm()** in the **DeviceManagerUiLibConstructor()** prevents from initiating references needed to navigate deeply within **UiApp.efi**.

Although CreateDeviceManagerForm is removed from DeviceManagerUiLibConstructor, it will still be called when opening `Device Manager` form. 
I noticed that your UEFI firmware is provided by AMI, which may carry out some customization based on edk2. If this issue is only reproduced on AMI firmware instead of edk2, I think it should not be a edk2 bug.