tianocore/edk2 #10952: Tails ISO images cannot be booted nor mounted in UEFI shell

davidebeatrici, 2025-04-13:
Every time QEMU and related packages get updated in OpenMandriva I test a few selected images to ensure there are no issues.

A few months ago I noticed all ISOs were still booting just fine with the exception of Tails', which are just not recognized at all. In the meantime QEMU got updated at least once more, but the issue has been persisting.

A brief investigation revealed that the image's filesystem cannot be mounted in the UEFI shell nor the setup when adding a new boot entry.

[image]

```
Disk tails-amd64-6.14.1.iso: 1.47 GiB, 1579253760 bytes, 3084480 sectors
Units: sectors of 1 * 512 = 512 bytes
Sector size (logical/physical): 512 bytes / 512 bytes
I/O size (minimum/optimal): 512 bytes / 512 bytes
Disklabel type: dos
Disk identifier: 0x0000002a

Device                  Boot Start     End Sectors  Size Id Type
tails-amd64-6.14.1.iso1 *        0 3084479 3084480  1.5G 17 Hidden HPFS/NTFS
```

For comparison, an image that boots just fine:

[image]

```
Disk kali-linux-2024.2-live-everything-amd64.iso: 12.56 GiB, 13486344192 bytes, 26340516 sectors
Units: sectors of 1 * 512 = 512 bytes
Sector size (logical/physical): 512 bytes / 512 bytes
I/O size (minimum/optimal): 512 bytes / 512 bytes
Disklabel type: dos
Disk identifier: 0xbc040884

Device                                       Boot    Start      End  Sectors  Size Id Type
kali-linux-2024.2-live-everything-amd64.iso1 *          64 26337867 26337804 12.6G 17 Hidden HPFS/NTFS
kali-linux-2024.2-live-everything-amd64.iso2      26337868 26339915     2048    1M  1 FAT12
```

Related: https://github.com/system76/firmware-open/issues/205

ajfish, 2025-04-14:
> davidebeatrici
>  created an issue
> Related: system76/firmware-open#205
> Related: system76/firmware-open#205 <https://github.com/system76/firmware-open/issues/205>

UEFI does not boot from generic ISO-9660 [1] it uses the “El Torito” Bootable CD-ROM Format Specification[2]. So I’d guess it might be a mastering issue with the ISO? The 1st thing to do is to check that sector 0x11 (17) contains a valid Boot Record as required by the  “El Torito” Specification.
'
[1] https://uefi.org/specs/UEFI/2.11/13_Protocols_Media_Access.html#cd-rom-and-dvd-rom
[2] https://pdos.csail.mit.edu/6.828/2017/readings/boot-cdrom.pdf

Thanks,

Andrew Fish

ajfish, 2025-04-14:
The UEFI code lives here: https://github.com/tianocore/edk2/blob/master/MdeModulePkg/Universal/Disk/PartitionDxe/ElTorito.c#L70

Thanks,

Andrew Fish

davidebeatrici, 2025-04-14:
Thank you.

```
$ isoinfo -d -i tails-amd64-6.14.1.iso 
Setting input-charset to 'UTF-8' from locale.
CD-ROM is in ISO 9660 format
System id: 
Volume id: TAILS 6.14.1 - 20250402
Volume set id: 
Publisher id: HTTPS://TAILS.NET/
Data preparer id: LIVE-BUILD 2.0.12-2; HTTP://PACKAGES.QA.DEBIAN.ORG/LIVE-BUILD
Application id: TAILS
Copyright File id: 
Abstract File id: 
Bibliographic File id: 
Volume set size is: 1
Volume set sequence number is: 1
Logical block size is: 2048
Volume size is: 769299
El Torito VD version 1 found, boot catalog is in sector 101

Joliet with UCS level 3 found.
SUSP signatures version 1 found
Rock Ridge signatures version 1 found
Rock Ridge id 'RRIP_1991A'
Eltorito validation header:
    Hid 1
    Arch 0 (x86)
    ID ''
    Cksum AA 55 OK
    Key 55 AA
    Eltorito defaultboot header:
        Bootid 88 (bootable)
        Boot media 0 (No Emulation Boot)
        Load segment 0
        Sys type 0
        Nsect 4
        Bootoff 66 102
```

```
$ isoinfo -d -i kali-linux-2024.2-live-everything-amd64.iso 
Setting input-charset to 'UTF-8' from locale.
CD-ROM is in ISO 9660 format
System id: 
Volume id: Kali Live
Volume set id: 
Publisher id: KALI
Data preparer id: LIVE-BUILD 20230502+KALI3; HTTPS://SALSA.DEBIAN.ORG/LIVE-TEAM/LIVE-BUILD
Application id: KALI LINUX
Copyright File id: 
Abstract File id: 
Bibliographic File id: 
Volume set size is: 1
Volume set sequence number is: 1
Logical block size is: 2048
Volume size is: 6585129
El Torito VD version 1 found, boot catalog is in sector 2038

Joliet with UCS level 3 found.
SUSP signatures version 1 found
Rock Ridge signatures version 1 found
Rock Ridge id 'RRIP_1991A'
Eltorito validation header:
    Hid 1
    Arch 0 (x86)
    ID ''
    Cksum AA 55 OK
    Key 55 AA
    Eltorito defaultboot header:
        Bootid 88 (bootable)
        Boot media 0 (No Emulation Boot)
        Load segment 0
        Sys type 0
        Nsect 4
        Bootoff 9F7 2551
```

Please note that the boot issue appeared suddenly without any changes on my side (with the exception of QEMU updates). Until a few months ago Tails ISOs could be booted just fine in UEFI mode.

davidebeatrici, 2025-05-06:
I have found another distribution with the same issue: [Slax](https://www.slax.org)

@Tomas-M

lersek, 2025-05-28:
@davidebeatrici

> Does EDK2 have CSM

It used to (and see `config CSM` in SeaBIOS's `src/Kconfig`), but over time, the edk2 CSM was relegated to OVMF, and we finally removed the CSM from OVMF too; see https://github.com/tianocore/edk2/issues/9435. (Some final loose threads were left to be tied up in https://github.com/tianocore/edk2/issues/9719.) If you check out edk2 at 238690a30d02 (or at `edk2-stable202311`, for a released tag), you might be able to build OVMF with the SeaBIOS CSM.

For general edk2 support, the most relevant historical tickets (now) seem to be https://github.com/tianocore/edk2/issues/7758 and https://github.com/tianocore/edk2/issues/7956. I believe the commit at which you'd need to check the tree out, to be ahead of the series of both of those tickets, is ddb8cedce7e0.

davidebeatrici, 2025-05-28:
Oh, makes sense! I believe I was using an OVMF version that still had CSM last time I successfully booted Tails on a QEMU UEFI machine.

In any case, what matters is the official confirmation this is not an issue in EDK2 but rather the ISOs (also verified on a physical machine).

For further reference:

https://tails.net/contribute/design/UEFI/
https://gitlab.tails.boum.org/tails/blueprints/-/wikis/UEFI

closed 2025-05-28 by davidebeatrici