tianocore/edk2 #10973: [Bug]: <title> Build target [DEBUG] is not supported by the platform for RK3566

GAVSONCHA, 2025-04-16:
### Is there an existing issue for this?

- [x] I have searched existing issues

### Bug Type

- [ ] Firmware
- [x] Tool
- [x] Unit Test

### What packages are impacted?

Other

### Which targets are impacted by this bug?

DEBUG, RELEASE

### Current Behavior

~/edk2/edk2$ build -p Platform/RK3566/RK3566.dsc -a AARCH64 -t GCC -b DEBUG
Build environment: Linux-5.15.167.4-microsoft-standard-WSL2-x86_64-with-glibc2.39
Build start time: 16:36:53, Apr.16 2025

WORKSPACE        = /home/gavso/edk2/edk2
PACKAGES_PATH    = /home/gavso/edk2/edk2
EDK_TOOLS_PATH   = /home/gavso/edk2/edk2/BaseTools
CONF_PATH        = /home/gavso/edk2/edk2/Conf
PYTHON_COMMAND   = python3

build.py...
 : error 2000: Invalid parameter
        Build target [DEBUG] is not supported by the platform. [Valid target: DEBUG RELEASE]

- Failed -
Build end time: 16:36:53, Apr.16 2025
Build total time: 00:00:00

### Expected Behavior

The build system should recognize DEBUG as a valid build target for the RK3566 platform, as specified in the BUILD_TARGETS field of the RK3566.dsc file (BUILD_TARGETS = DEBUG RELEASE).
Upon running the build command:

build -p Platform/RK3566/RK3566.dsc -a AARCH64 -t GCC -b DEBUG -n 4

it is expected that the build process starts, compiles all necessary components (e.g., HelloWorld.inf), and creates the Build/RK3566/DEBUG_* output directory with build logs and binaries.

### Steps To Reproduce

1) git clone https://github.com/tianocore/edk2.git
cd edk2
2)export WORKSPACE=$PWD
export EDK_TOOLS_PATH=$WORKSPACE/BaseTools
export PACKAGES_PATH=$WORKSPACE
export GCC_AARCH64_PREFIX=/usr/bin/aarch64-linux-gnu-
3) create file RK3566.dsc
4)build -p Platform/RK3566/RK3566.dsc -a AARCH64 -t GCC -b DEBUG -n 4

### Build Environment

```markdown
Ubuntu 20.04 в WSL2 (Windows Subsystem for Linux)
```

### Version Information

```text
latest
```

### Urgency

High

### Are you going to fix this?

I will fix it

### Do you need maintainer feedback?

No maintainer feedback needed

### Anything else?

file RK3566.dsc: 
[Defines]
  PLATFORM_NAME           = RK3566
  PLATFORM_GUID           = {6b1a6c44-c9f7-4d0e-bf17-cf415493544f}
  PLATFORM_VERSION        = 1.0
  DSC_SPECIFICATION       = 0x00010005
  OUTPUT_DIRECTORY        = Build/RK3566
  SUPPORTED_ARCHITECTURES = AARCH64
  BUILD_TARGETS           = DEBUG RELEASE
  SKUID_IDENTIFIER        = DEFAULT
  FLASH_DEFINITION        = Platform/RK3566/RK3566.fdf

[LibraryClasses]
  UefiDriverEntryPoint|MdePkg/Library/UefiDriverEntryPoint/UefiDriverEntryPoint.inf
  UefiBootServicesTableLib|MdePkg/Library/UefiBootServicesTableLib/UefiBootServicesTableLib.inf
  UefiLib|MdePkg/Library/UefiLib/UefiLib.inf

[Components]
  /home/gavso/edk2/edk2/HelloWorld/HelloWorld.inf

EricGao2015, 2025-04-17:
`BUILD_TARGETS = DEBUG|RELEASE` is right usage.

lgao4, 2025-05-29:
Please go ahead.