tianocore/edk2 #11002: Default Seed used for random number generator in CryptoPkg.

kanagavels97, 2025-04-24:
Hi @mdkinney 

The random number generators must be seeded with unpredictable values before performing any cryptographic operations. Operations using a default random seed can be compromised/bypassed by an attacker.

The implementation for RandomSeed is as follows, where a NULL value for the Seed parameter results in the usage of DefaultSeed.

[image]

The following functionality is present in CryptPkcs7Sign.c file, which is an OpenSSL wrapper where the default seed is used before generating a private key used for signing.

[image]
[image]

The following code snippet is in TlsInit.c file for initializing TLS functionality, which uses the default seed value for the random number generator.

[image]

Using a known or predictable seed value weakens or fully compromises the cryptographic operation, allowing it to be exploited. An attacker can bypass TLS session authentication or mechanisms validating PKCS #7 signatures.

Could you explain why the RandomSeed() function is called with a seed value of NULL and a seed size of 0?

Could you let us know if there are plans to implement a Cryptographically Secure Pseudo-Random Number Generator (CSPRNG)?

Thanks,
Kanagavel

cc: srinivasanmani@ami.com gayathrit@ami.com prarthanasagar@ami.com kanagavels@ami.com

liyi77, 2025-04-30:
For X86, we default to use tsc as random seed (RandTsc.c).

Even so its entropy is not good enough for cryptography.
Using RngLib instead might be a better approach.
@jyao1

kanagavels97, 2025-05-30:
Hi @liyi77 @mdkinney 

In the CryptRandTsc.file, the RandomSeed function uses the current system time as a seed when it is called with a NULL seed value and a seed size of 0. However, since the current time is predictable, it does not provide strong randomness.

Based on your recommendation, we have decided to use RngLib as the random number generator for both x64 and AArch64 architectures.

Once we’ve completed the implementation using RngLib, would it be appropriate to submit a patch to EDK II? We’d appreciate your guidance or suggestions on this.

Thanks,
Kanagavel