tianocore/edk2 #11035: [Bug]: [OvmfPkg] Got "Invalid MMIO opcode (0)" in ovmf  when installing a sev guest from CD-ROM

r1chard-lyu, 2025-05-05:
### Is there an existing issue for this?

- [x] I have searched existing issues

### Bug Type

- [x] Firmware
- [ ] Tool
- [ ] Unit Test

### What packages are impacted?

OvmfPkg

### Which targets are impacted by this bug?

DEBUG

### Current Behavior

When installing a SEV guest on a SEV-enabled host using a CD-ROM as the installation source, the following error is triggered:
```
Invalid MMIO opcode (0)
ASSERT CcExitVcHandler.c(498): ((BOOLEAN)(0==1))
```

Upon inspecting the source code at CcExitVcHandler.c:498, it appears the assertion is caused by an invalid MMIO opcode (0x00) issued by the guest VM:
```
switch (OpCode) {
  ...
  case 0xBE:
  ...
  case 0xBF:
  ...
  default:
    DEBUG ((DEBUG_ERROR, "Invalid MMIO opcode (%x)\n", OpCode));
    Status = GP_EXCEPTION;
    ASSERT (FALSE);
```

This indicates that the MMIO exit handler does not properly handle unrecognized or invalid opcodes, which causes the guest to hang or crash during installation.

### Expected Behavior

The MMIO exit should be able to handle invalid MMIO opcodes. For example, this issue fails to handle the 0x00 opcode, which leads to a hang that should be avoided.

### Steps To Reproduce

I triggered this bug on a SLES15-SP6 host while installing a SLES15-SP6 SEV guest using an ISO file from the CD-ROM, with the following installation command.

virt-install --name sles15sp7-sev-es --cdrom /virt1-share-iso/sles15sp6/SLE-15-SP6-Full-x86_64-Media1.iso --disk size=20 --network=bridge=br0,model=virtio,rom.bar=off  --vcpus 4 --memory 4096 --events on_reboot=destroy --machine q35 --memtune hard_limit=4563402 --launchSecurity sev,policy=0x07  --boot firmware=efi,loader.stateless=yes  --console pty,target.type=serial --graphics none --serial pty --qemu-commandline="-global isa-debugcon.iobase=0x402 -debugcon file:/tmp/ovmf-debug.log"

### Build Environment

```markdown
- OS(s): SLES15-SP6
- Tool Chain(s): GCC5
```

### Version Information

```text
edk2-stable202502
```

### Urgency

Medium

### Are you going to fix this?

I will fix it

### Do you need maintainer feedback?

Maintainer feedback requested

### Anything else?

I'm willing to help resolve this issue, but I would appreciate any suggestions or feedback.

lgao4, 2025-05-29:
Please go ahead.

r1chard-lyu, 2025-06-21:
Thank you for all your suggestions. Yes, I should propose using this issue to reproduce the problem on the public distro openSUSE 15.6, as @jfehlig mentioned. For now, I am still working on this issue.

r1chard-lyu, 2025-06-23:
VMEXIT is divided into AE (Automatic Exit) events and NAE (Non-Automatic Exit) events.
MMIO should be an NAE event, but the -cdrom parameter causes it to be handled as an AE event. Consequently, the GHCB is not synchronized and is empty.