tianocore/edk2 #11038: [Bug]: <title>Smm data might get leaked via S3SaveState Protocol

wjqsec, 2025-05-05:
### Is there an existing issue for this?

- [x] I have searched existing issues

### Bug Type

- [x] Firmware
- [ ] Tool
- [ ] Unit Test

### What packages are impacted?

MdePkg

### Which targets are impacted by this bug?

RELEASE, NOOPT, NO-TARGET, DEBUG

### Current Behavior

Hello,

I would like to discuss a potential System Management Mode (SMM) memory data leakage problem in the current edk2 implementation.

In edk2/MdeModulePkg/Universal/Acpi/SmmS3SaveState/SmmS3SaveState.c, it defines a S3SmmSaveStateProtocol, which is used for storing data in the SMRAM and restoring it during the system resuming process. However, it does not check if the pointer that points to the data is exactly from the outside SMRAM. A malicious attacker might craft a pointer that points to SMRAM. The protocol simply trusts the pointer and performs a CopyMem on the data and restores it to the user-specified address.

### Expected Behavior

A malicious attacker can leak SMRAM data to Non-SMRAM.

### Steps To Reproduce

1. Locate the EFI_S3_SMM_SAVE_STATE_PROTOCOL. 
2. Manually craft a pointer that points to SMRAM.
3. Invoke the Protocol's BootScriptWrite function and pass the crafted pointer as an argument.

### Build Environment

```markdown
- OS(s):Ubuntu 20.04
- Tool Chain(s):GCC5
```

### Version Information

```text
edk2/master
```

### Urgency

Medium

### Are you going to fix this?

Someone else needs to fix it

### Do you need maintainer feedback?

Maintainer feedback requested

jkmathews, 2025-05-05:
Thank you for the report. Please give us a few a few days to review and respond.

In the future, please follow this process for reporting potential security vulnerabilities:
https://github.com/tianocore/tianocore.github.io/wiki/reporting-security-issues

wjqsec, 2025-05-05:
Hi, I am not sure yet if it is really a security vulnerability or it is just my misunderstandings to the code. So I opened a discussion here.

jkmathews, 2025-05-07:
It does not look like this is a security vulnerability. 
- EFI_S3_SMM_SAVE_STATE_PROTOCOL is only available in SMM. Exploit would require somehow executing untrusted code in SMM to locate & misuse the protocol. 
- With the DXE version of the protocol, EFI_S3_SAVE_STATE_PROTOCOL, updates are not supported after EndOfDxe event. After EndOfDxe only SMM can add to the LockBox / BootScript.
- During S3 resume, SMRAM is closed and locked prior to executing the BootScript.

While we don't see any exploit, we think it is a good suggestion to add bounds checks to this protocol.  It's good coding practice for defense in depth.

wjqsec, 2025-05-07:
Hi,

Thanks a lot for the response.

Yes you are right that only SMM EFI_S3_SMM_SAVE_STATE_PROTOCOL can access SMRAM after Lockbox event . However, I have seen that some vendor UEFI firmware misuse this protocol to store data. What they do is:
    1. Register a SMI handler which is accessible even after Lockbox.
    2. This handler locates EFI_S3_SMM_SAVE_STATE_PROTOCOL to store data by passing user-controllable data as arguments (and ofc user-provided  pointers).
    3. The vendor customized SMI handler does not check any user-provided pointers and directly pass them to the protocol.

This is why I proposed this discussion. I thought checking the data inside the protocol implementation would be a better way. What do you think?

lgao4, 2025-05-29:
Can you share your proposal by creating Pull Request?

wjqsec, 2025-05-30:
sure, will do