tianocore/edk2 #11040: [CryptoPkg]Disabled Anti replay and Hello verify in MbedTls configuration.

kanagavels97, 2025-05-06:
Hi @mdkinney 

The **MBEDTLS_SSL_DTLS_ANTI_REPLAY** and **MBEDTLS_SSL_DTLS_HELLO_VERIFY** are highly recommended for DTLS servers to enable. These are disabled in our CryptoPkg.

I can see these are enabled in mbedtls_config.h file from the mbedtls repository.

The cipher suites below have security risks and are recommended to be disabled, but these are enabled in our CryptoPkg.

**MBEDTLS_KEY_EXCHANGE_DHE_PSK_ENABLED
MBEDTLS_KEY_EXCHANGE_DHE_RSA_ENABLED
MBEDTLS_SSL_RENEGOTIATION**

Could you please let us know if you have any specific reasons to enable the above mentioned ciphersuites?

Could you please let us know if ypu have any specific reasons to disable **MBEDTLS_SSL_DTLS_ANTI_REPLAY** and **MBEDTLS_SSL_DTLS_HELLO_VERIFY** ?

Thanks,
Kanagavel

cc: @srinivasanm@ami.com @gayathrit@ami.com @prarthanasv@ami.com @kanagavels@ami.com

liyi77, 2025-05-12:
These configs are related to TLS.
TLS of MbedTls is not enabled in edk2, we only use the crypto part so this is not an issue.

kanagavels97, 2025-05-19:
Hi @liyi77 

I have checked the implementation and can see the replacement for the following algorithms.
MBEDTLS_KEY_EXCHANGE_DHE_PSK_ENABLED
MBEDTLS_KEY_EXCHANGE_DHE_RSA_ENABLED
MBEDTLS_SSL_RENEGOTIATION

Can I comment out the above mentioned and enable the MBEDTLS_SSL_DTLS_ANTI_REPLAY and MBEDTLS_SSL_DTLS_HELLO_VERIFY and submit the patch?

Please provide your suggestions.

Thanks,
Kanagavel

liyi77, 2025-05-19:
> Can I comment out the above mentioned and enable the MBEDTLS_SSL_DTLS_ANTI_REPLAY and MBEDTLS_SSL_DTLS_HELLO_VERIFY and submit the patch?
> 
Sure, please take it.

liyi77, 2026-09-15:
https://github.com/tianocore/edk2/pull/11211

closed 2026-09-15 by liyi77