tianocore/edk2 #11044: [Code First]: Add EFI_MEMORY_SHARED memory attribute

deeglaze, 2025-05-06:
### Code First Item Overview

## Status:

Submitted to industry standard forum

Discussed at a forum meeting, and there were no objections.

## Document:

UEFI Specification Version 2.11

## License

SPDX-License-Identifier: CC-BY-4.0

## Submitter: [TianoCore Community](https://www.tianocore.org)

## Summary of the change

Adds a `#define` to UefiSpec.h for use GetMemoryMap's memory attributes.

## Benefits of the change

Permits runtime services to use memory allocated for DMA by directing the guest OS to map it appropriately. 

## Impact of the change

For Confidential Computing, this means that runtime services can use shared (unencrypted) memory to communicate with the hypervisor with higher throughput than MMIO/PIO.

## Detailed description of the change [normative updates]

Add `#define EFI_MEMORY_SHARED          0x0000000000200000` to UefiSpec.h

## Special Instructions

For adopters of this attribute to existing runtime services, it is recommended to perform a page table walk to determine if the guest OS was enlightened to the memory attribute to map memory correctly. If not, to fall back on MMIO/PIO.

### What specification(s) are directly related?

UEFI

deeglaze, 2025-05-06:
/cc @jyao1

deeglaze, 2025-05-06:
Given https://github.com/tianocore/edk2/pull/10695 has merged, @kraxel this is relevant to the runtime behavior for the OVMF variable service.

kraxel, 2025-05-07:
> Given [#10695](https://github.com/tianocore/edk2/pull/10695) has merged, [@kraxel](https://github.com/kraxel) this is relevant to the runtime behavior for the OVMF variable service.
> 
> Looks like `mCommunicateBuffer` is what would be the runtime pages we want to allocate with the EFI_MEMORY_SHARED attribute for confidential computing.

Well, technically yes.  Question is though how useful it is to trust the host with the EFI variables in a confidential computing setting ...

I'm looking into running the EFI variable service in svsm instead.  `VirtMmCommunicationDxe` would call a svsm protocol instead of talking to qemu then, and the host does not need access to `mCommunicateBuffer` any more.

jyao1, 2025-05-08:
To clarify: The usage of this EFI_MEMORY_SHARED bit is that, when the firmware boots to OS,
 * if the firmware has NO DMA buffer present, then no buffer should have this BIT. (That is current status in EDKII.)
 * if the firmware has shared DMA buffer present, then this shared DMA buffer should have this BIT. (It depends on the new use case.)
 * if the firmware has private DMA buffer present, then this private DMA buffer should NOT have this BIT. (Maybe for future if we have TEE-IO device enabled in firmware phase)

Is that right understanding?

deeglaze, 2025-05-08:
> To clarify: The usage of this EFI_MEMORY_SHARED bit is that, when the firmware boots to OS,
>
> * if the firmware has NO DMA buffer present, then no buffer should have this BIT. (That is current status [...]

This is what I would expect, however I did not propose the generalization to be any DMA–just the notion of shared memory. For memory that's not shared but should be mapped in vIOMMU, that seems more appropriate for a different memory attribute. @ardbiesheuvel what are your thoughts?

deeglaze, 2025-06-09:
Checking back in. @ardbiesheuvel ?

ardbiesheuvel, 2025-06-13:
> To clarify: The usage of this EFI_MEMORY_SHARED bit is that, when the firmware boots to OS,
>
> * if the firmware has NO DMA buffer present, then no buffer should have this BIT. (That is current status [...]

Exactly.

And `firmware` in this context is the set of runtime DXE drivers that needs to rely on the OS to create the CPU side mappings of the EFI_MEMORY_RUNTIME regions. IOW, it does not make sense for EFI_MEMORY_SHARED to be set on regions that are not EFI_MEMORY_RUNTIME, as there are no situations where the firmware has to rely on mappings created by any other agent than itself.

ardbiesheuvel, 2025-06-13:
> This is what I would expect, however I did not propose the generalization to be any DMA–just the notion of shared memory. For memory that's not shared but should be mapped in vIOMMU, that seems more [...]

I suggested the generalization because I think there could be other cases where this could be useful, and limiting it specifically to virtual machines under confidential compute requires a number of new definitions in the common parts of the spec that don't exist today. If we can make the semantics clear and unambiguous, I don't think there is a need to only allow it in a very limited set of use cases.

jyao1, 2025-06-16:
Right. I agree to generalize it.
The DMA buffer usage I gave could be treated as the example to clarify the possible use case, but it does not mean the only use case.

jmarinho, 2025-06-18:
@deeglaze will the model allow EFI_MEMORY_SHARED memory to be mapped with execute permissions?
If that's not intended, should the spec language forbid it?

deeglaze, 2025-06-18:
I don't see why we should limit that, since it's threat model–specific.

jmarinho, 2025-06-18:
> I don't see why we should limit that, since it's threat model–specific.

Is the purpose of EFI_MEMORY_SHARED to enable the interaction between 2 parties (e.g. VMM and CC guest)?

You raise a good point that each entity can independently decide on the permissions to meet its security goals.

My question is: by explicitly forbidding execute permission, the spec sets clear boundaries on the interations between the 2 parties. Is that valuable for interoperability?

deeglaze, 2025-06-18:
I don't know if folks want to offload some JIT compilation weirdness to a device and need executable DMA. It shouldn't matter. If you don't trust your firmware to keep DMA non-executable, then don't use that firmware.