tianocore/edk2 #11050: Save and restore mm6 and mm7 registers in OpensslLibFullAccel.inf

PrarthanaSagarV, 2025-05-08:
In many of the ASM files in OpensslLibFullAccel.inf, mm6 and mm7 registers are used without saving and restoring the original values. Because of this, if any values are stored in mm6/m7 registers, it will be destroyed in OpensslLibFullAccel asm files. So, we faced hang issues.
As per [X64 ABI Convention](https://learn.microsoft.com/en-us/cpp/build/x64-software-conventions?view=msvc-170) it is required to restore mm6 and mm7 registers for both MSFT and GCC ASM files

[image]

PrarthanaSagarV, 2025-05-08:
@srinivasanm@ami.com @kanagavels97 @gayathrit@ami.com @karthikar@ami.com

mdkinney, 2025-05-09:
Is this for IA32 or X64 builds?

mdkinney, 2025-05-09:
Which specific X64 ASM files are not saving/restoring.  I am looking at a few and I see the save/restore of XMM6..XMM15.

For example:
Save: https://github.com/tianocore/edk2/blob/fc215474c48295608f3e2a4f227554b810ceb2bb/CryptoPkg/Library/OpensslLib/OpensslGen/X64-MSFT/crypto/aes/aesni-sha1-x86_64.nasm#L63
Restore: https://github.com/tianocore/edk2/blob/fc215474c48295608f3e2a4f227554b810ceb2bb/CryptoPkg/Library/OpensslLib/OpensslGen/X64-MSFT/crypto/aes/aesni-sha1-x86_64.nasm#L1402

PrarthanaSagarV, 2025-05-09:
Below are some of the files where the registers are not restored.

[OpensslGen/IA32-MSFT/crypto/sha/sha512-586.nasm](https://github.com/tianocore/edk2/blob/fc215474c48295608f3e2a4f227554b810ceb2bb/CryptoPkg/Library/OpensslLib/OpensslGen/IA32-MSFT/crypto/sha/sha512-586.nasm)
OpensslGen\X64-MSFT\crypto\ec\ecp_nistz256-x86_64.nasm
OpensslGen\IA32-MSFT\crypto\ec\ecp_nistz256-x86.nasm
OpensslGen\IA32-MSFT\crypto\bn\x86-mont.nasm
OpensslGen\IA32-MSFT\crypto\bn\x86-gf2m.nasm
OpensslGen\IA32-MSFT\crypto\bn\bn-586.nasm
OpensslGen\X64-MSFT\crypto\modes\ghash-x86_64.nasm
OpensslGen\X64-MSFT\crypto\modes\aesni-gcm-x86_64.nasm
OpensslGen\X64-MSFT\crypto\modes\aes-gcm-avx512.nasm
OpensslGen\X64-GCC\crypto\bn\rsaz-x86_64.s
OpensslGen\X64-GCC\crypto\ec\ecp_nistz256-x86_64.s