tianocore/edk2 #11084: [Bug]: Image type X64 can't be loaded on AARCH64 UEFI system

Libo-at-windriver, 2025-05-16:
### Is there an existing issue for this?

- [x] I have searched existing issues

### Bug Type

- [x] Firmware
- [ ] Tool
- [ ] Unit Test

### What packages are impacted?

ArmVirtPkg

### Which targets are impacted by this bug?

DEBUG

### Current Behavior

When launching a Qemu VM with edk2 in a Xilinx ZCU102 (AARCH64) board, the Qemu VM went to login prompt, however, the following error showed up.

	Image type X64 can't be loaded on AARCH64 UEFI system.

When enabling TPM2 (-D TPM2_ENABLE=TRUE -D TPM2_CONFIG_ENABLE=TRUE -D TPM1_ENABLE=TRUE), there are some errors below:
	Tpm2SubmitCommand - Tcg2 - Not Found
	Tpm2GetCapabilityPcrs fail!
	Tpm2SubmitCommand - Tcg2 - Not Found

### Expected Behavior

The error "Image type X64 can't be loaded on AARCH64 UEFI system" should not appear in an AARCH64 target board and Tpm2 should not get failed.

### Steps To Reproduce

1. git clone edk2

git clone https://github.com/tianocore/edk2.git
cd edk2
git submodule update --init
cd ..

2. use the document below to set up docker environment
https://github.com/tianocore/tianocore.github.io/wiki/How-to-Develop-With-Containers

3. build AARCH64 edk2
build -v -t GCC5 -p ArmVirtPkg/ArmVirtQemu.dsc --pcd PcdFirmwareVersionString=Ledk2-stable202302-for-qemu --pcd PcdFirmwareReleaseDateString=L03/01/2023 -n 12 -a AARCH64 -D NETWORK_HTTP_BOOT_ENABLE=TRUE -D NETWORK_IP6_ENABLE=TRUE -D NETWORK_TLS_ENABLE=FALSE -D NETWORK_ISCSI_ENABLE=TRUE -D NETWORK_ALLOW_HTTP_CONNECTIONS=TRUE -D TPM2_ENABLE=FALSE -D TPM2_CONFIG_ENABLE=FALSE -D TPM1_ENABLE=FALSE -D CAVIUM_ERRATUM_27456=FALSE -D DEBUG_PRINT_ERROR_LEVEL=0x80020048 --pcd PcdDxeNxMemoryProtectionPolicy=0xC000000000007FD1 --pcd PcdUninstallMemAttrProtocol=TRUE -b DEBUG

4. run Qemu in Xilinx ZCU102 board
4.1 ssh to the target Xilinx ZCU102 board

4.2 run the following Qemu command

/usr/bin/qemu-system-aarch64 \
 -enable-kvm \
 -m size=2097152k \
 -cpu host \
 -M virt \
 -nographic \
 -drive if=pflash,format=raw,file=/usr/share/qemu/edk2-aarch64-code.fd,readonly=on \
 -drive if=pflash,format=raw,file=/usr/share/qemu/edk2-arm-vars.fd \
 -drive if=none,file=/opt/debian-11-nocloud-arm64.qcow2,id=hd0 \
 -device virtio-blk-device,drive=hd0

### Build Environment

```markdown
- OS(s):Linux amd-zynqmp 6.6.87-yocto-standard #1 SMP PREEMPT Mon Apr 14 03:22:16 UTC 2025 aarch64 GNU/Linux
- Tool Chain(s):GCC5
```

### Version Information

```text
Commit: b2db39cefe1318d13d30040d31a6954e6b7219ce
```

### Urgency

Medium

### Are you going to fix this?

Someone else needs to fix it

### Do you need maintainer feedback?

Maintainer feedback requested

### Anything else?

I'm not sure if the following errors are issue. Here's my understanding: Tpm2 is software-emulated in edk2 and it has been successfully compiled for AARCH64 and it should be working here instead of complaining "Tpm2GetCapabilityPcrs fail!". Please correct me if there is any misunderstanding.

	Tpm2SubmitCommand - Tcg2 - Not Found
	Tpm2GetCapabilityPcrs fail!
	Tpm2SubmitCommand - Tcg2 - Not Found

Libo-at-windriver, 2025-05-16:
Here's the log. The line "=== Image->ImageContext.Machine=X64, status=0, type=B, size=683008, PdbPointer=<null string>, Info.FilePath=" is the debugging information that I added in CoreLoadPeImage(). What's confusing me is that I don't know which image caused the issue since the PdbPointer is null.

add-symbol-file /buildarea5/xyz/edk2/Build/ArmVirtQemu-AARCH64/DEBUG_GCC5/AARCH64/OvmfPkg/AcpiPlatformDxe/AcpiPlatformDxe/DEBUG/QemuFwCfgAcpiPlatform.dll 0xBEF95000
Loading driver at 0x000BEF94000 EntryPoint=0x000BEF995D8 QemuFwCfgAcpiPlatform.efi
InstallProtocolInterface: BC62157E-3E33-4FEC-9920-2D3B36D750DF BF0A1C98
ProtectUefiImageCommon - 0xBF0A1040
  - 0x00000000BEF94000 - 0x000000000000A000
Found FwCfg @ 0x9020008/0x9020000
Found FwCfg DMA @ 0x9020010
AcpiPlatformEntryPoint: waiting for root bridges to be connected, registered callback
[Bds] Entry...
[BdsDxe] Locate Variable Policy protocol - Success
Variable Driver Auto Update Lang, Lang:eng, PlatformLang:en Status: Success
DXE - Total Runtime Image Count: 0x8
DXE - Dump Runtime Image Records:
RealTimeClock.efi: 0xBF0F0000 - 0xBF120000
  Code Section: 0xBF100000 - 0xBF110000
MonotonicCounterRuntimeDxe.efi: 0xBF130000 - 0xBF160000
  Code Section: 0xBF140000 - 0xBF150000
VariableRuntimeDxe.efi: 0xBF280000 - 0xBF2C0000
  Code Section: 0xBF290000 - 0xBF2A0000
VirtNorFlashDxe.efi: 0xBF3E0000 - 0xBF420000
  Code Section: 0xBF3F0000 - 0xBF400000
ResetSystemRuntimeDxe.efi: 0xBF480000 - 0xBF4C0000
  Code Section: 0xBF490000 - 0xBF4A0000
ReportStatusCodeRouterRuntimeDxe.efi: 0xBF4D0000 - 0xBF510000
  Code Section: 0xBF4E0000 - 0xBF4F0000
CapsuleRuntimeDxe.efi: 0xBFE20000 - 0xBFE50000
  Code Section: 0xBFE30000 - 0xBFE40000
RuntimeDxe.efi: 0xBFE60000 - 0xBFEA0000
  Code Section: 0xBFE70000 - 0xBFE80000
[Variable]END_OF_DXE is signaled
Initialize variable error flag (FF)
PCI Bus First Scanning
PciBus: Discovered PCI @ [00|00|00]  [VID = 0x1B36, DID = 0x8]

PciBus: Discovered PCI @ [00|01|00]  [VID = 0x1AF4, DID = 0x1000]
   BAR[0]: Type =   Io32; Alignment = 0x1F;     Length = 0x20;  Offset = 0x10
   BAR[1]: Type =  Mem32; Alignment = 0xFFF;    Length = 0x1000;        Offset = 0x14
   BAR[4]: Type = PMem64; Alignment = 0x3FFF;   Length = 0x4000;        Offset = 0x20

PCI Bus Second Scanning
PciBus: Discovered PCI @ [00|00|00]  [VID = 0x1B36, DID = 0x8]

PciBus: Discovered PCI @ [00|01|00]  [VID = 0x1AF4, DID = 0x1000]
   BAR[0]: Type =   Io32; Alignment = 0x1F;     Length = 0x20;  Offset = 0x10
   BAR[1]: Type =  Mem32; Alignment = 0xFFF;    Length = 0x1000;        Offset = 0x14
   BAR[4]: Type = PMem64; Alignment = 0x3FFF;   Length = 0x4000;        Offset = 0x20

PciBus: Discovered PCI @ [00|00|00]  [VID = 0x1B36, DID = 0x8]

PciBus: Discovered PCI @ [00|01|00]  [VID = 0x1AF4, DID = 0x1000]
   BAR[0]: Type =   Io32; Alignment = 0x1F;     Length = 0x20;  Offset = 0x10
   BAR[1]: Type =  Mem32; Alignment = 0xFFF;    Length = 0x1000;        Offset = 0x14
   BAR[4]: Type = PMem64; Alignment = 0x3FFF;   Length = 0x4000;        Offset = 0x20

PciHostBridge: SubmitResources for PciRoot(0x0)
 I/O: Granularity/SpecificFlag = 0 / 01
      Length/Alignment = 0x1000 / 0xFFF
 Mem: Granularity/SpecificFlag = 32 / 00
      Length/Alignment = 0x100000 / 0xFFFFF
 Mem: Granularity/SpecificFlag = 64 / 00
      Length/Alignment = 0x100000 / 0xFFFFF
PciBus: HostBridge->SubmitResources() - Success
PciHostBridge: NotifyPhase (AllocateResources)
 RootBridge: PciRoot(0x0)
  Mem64: Base/Length/Alignment = 8000000000/100000/FFFFF - Success
  Mem: Base/Length/Alignment = 10000000/100000/FFFFF - Success
  I/O: Base/Length/Alignment = 0/1000/FFF - Success
PciBus: HostBridge->NotifyPhase(AllocateResources) - Success
Process Option ROM: BAR Base/Length = 10000000/40000
PciBus: Resource Map for Root Bridge PciRoot(0x0)
Type =   Io16; Base = 0x0;      Length = 0x1000;        Alignment = 0xFFF
   Base = 0x0;  Length = 0x20;  Alignment = 0x1F;       Owner = PCI [00|01|00:10]
Type =  Mem32; Base = 0x10000000;       Length = 0x100000;      Alignment = 0xFFFFF
   Base = 0x10000000;   Length = 0x40000;       Alignment = 0x3FFFF;    Owner = PCI [00|00|00:00]; Type =  OpRom
   Base = 0x10040000;   Length = 0x1000;        Alignment = 0xFFF;      Owner = PCI [00|01|00:14]
Type =  Mem64; Base = 0x8000000000;     Length = 0x100000;      Alignment = 0xFFFFF
   Base = 0x8000000000; Length = 0x4000;        Alignment = 0x3FFF;     Owner = PCI [00|01|00:20]; Type = PMem64

InstallProtocolInterface: 09576E91-6D3F-11D2-8E39-00A0C969723B BF043E98
InstallProtocolInterface: 4CF5B200-68B8-4CA5-9EEC-B23E3F50029A BF0A0428
InstallProtocolInterface: 09576E91-6D3F-11D2-8E39-00A0C969723B BF043F18
InstallProtocolInterface: 4CF5B200-68B8-4CA5-9EEC-B23E3F50029A BF0A0828
InstallProtocolInterface: 4006C0C1-FCB3-403E-996D-4A6C8724E06D BF0A08F0
[Security] 3rd party image[0] can be loaded after EndOfDxe: PciRoot(0x0)/Pci(0x1,0x0)/Offset(0x10E00,0x273FF).
InstallProtocolInterface: 5B1B31A1-9562-11D2-8E3F-00A0C969723B BF09F8C0
=== Image->ImageContext.Machine=X64, status=0, type=B, size=683008, PdbPointer=<null string>, Info.FilePath=
Image type X64 can't be loaded on AARCH64 UEFI system.
Unloading driver at 0x00000000000
InstallProtocolInterface: 3BC1B285-8A15-4A82-AABF-4D7D13FB3265 BF0A08D8
InstallProtocolInterface: 30CFE3E7-3DE1-4586-BE20-DEABA1B3B793 0
OnRootBridgesConnected: root bridges have been connected, installing ACPI tables
InstallProtocolInterface: 928939B2-4235-462F-9580-F6A2B2C21A4F 0
InstallQemuFwCfgTables: installed 9 tables

Libo-at-windriver, 2025-05-16:
BTW, I also used the UEFITool (https://github.com/LongSoft/UEFITool/releases/tag/A71) to examine the file edk2-aarch64-code.fd and found that the Machine Type of each file inside the fd file was AArch64. So I wonder which file caused "Image type X64"?

lersek, 2025-05-27:
I think using libvirt is by far the easiest for configuring swtpm. I don't think I've ever used swtpm manually, so I can only give you results from a cursory web search:
- https://libvirt.org/formatdomain.html#tpm-device
- https://qemu-project.gitlab.io/qemu/specs/tpm.html
- https://github.com/tompreston/qemu-ovmf-swtpm