tianocore/edk2 #11128: [Bug]: Incorrect array length for pcd values in auto-generated code

klassiker, 2025-05-28:
### Is there an existing issue for this?

- [x] I have searched existing issues

### Bug Type

- [ ] Firmware
- [x] Tool
- [ ] Unit Test

### What packages are impacted?

BaseTools

### Which targets are impacted by this bug?

RELEASE

### Current Behavior

The hash of `IScsiDxe.efi` changes depending on `MAX_CONCURRENT_THREAD_NUMBER` when building `OvmfPkg/OvmfPkgX64.dsc`.

The only change I could see / read is in `Build/OvmfX64/RELEASE_GCC5/X64/NetworkPkg/IScsiDxe/IScsiDxe/DEBUG/AutoGen.c`:

```c
$ grep 'UINT8 _gPcd_FixedAtBuild_PcdEntryPointOverrideDefaultValue' Build/OvmfX64/RELEASE_GCC5/X64/NetworkPkg/IScsiDxe/IScsiDxe/DEBUG/AutoGen.c
GLOBAL_REMOVE_IF_UNREFERENCED const UINT8 _gPcd_FixedAtBuild_PcdEntryPointOverrideDefaultValue[4] = {110, 111, 0 };
extern const UINT8 _gPcd_FixedAtBuild_PcdEntryPointOverrideDefaultValue[4];
```

This was only observed for `PcdEntryPointOverrideDefaultValue`.

### Expected Behavior

Auto-generated code does not change dependent on thread count.
Array length matches the actual value.

### Steps To Reproduce

```sh
mkdir bug && cd bug

git clone --single-branch --depth=1 https://github.com/tianocore/edk2 .

git submodule update --depth 1 --init MdePkg/Library/MipiSysTLib/mipisyst
git submodule update --depth 1 --init MdeModulePkg/Library/BrotliCustomDecompressLib/brotli
git submodule update --depth 1 --init CryptoPkg/Library/OpensslLib/openssl
git submodule update --depth 1 --init CryptoPkg/Library/MbedTlsLib/mbedtls
git submodule update --depth 1 --init SecurityPkg/DeviceSecurity/SpdmLib/libspdm
git submodule update --depth 1 --init BaseTools/Source/C/BrotliCompress/brotli

source ./edksetup.sh
make -C BaseTools ARCH=X64

build -q -s -a X64 -p OvmfPkg/OvmfPkgX64.dsc -b RELEASE -t GCC5 -n 1

grep 'UINT8 _gPcd_FixedAtBuild_PcdEntryPointOverrideDefaultValue' Build/OvmfX64/RELEASE_GCC5/X64/NetworkPkg/IScsiDxe/IScsiDxe/DEBUG/AutoGen.c
```

### Build Environment

```markdown
- OS(s): Ubuntu 24.04 (local), Fedora 42 (docker)
- Tool Chain(s): GCC5
- Docker: `fedora@sha256:ee88ab8a5c8bf78687ddcecadf824767e845adc19d8cdedb56f48521eb162b43`
```

### Version Information

```text
At least from `edk2-stable202502`, probably older

Also verified on current `master`:
Commit: 748bea71714a10caadcf457c90975312aaf6fb17
```

### Urgency

Low

### Are you going to fix this?

Someone else needs to fix it

### Do you need maintainer feedback?

No maintainer feedback needed

### Anything else?

Tried to debug it and found a race between `OvmfPkg/OvmfPkg.dec` and `OvmfPkg/Include/Dsc/NetworkComponents.dsc.inc`, where a Pcd instance might be reused for a different value or not copied correctly.

The size is explicitly not recalculated in `BaseTools/Source/Python/AutoGen/ModuleAutoGenHelper.py` and is a leftover from `OvmfPkg/OvmfPkg.dec` as it matches the length of the variable defined there.

This happens reliably for me on `edk2-stable202502` for two different machines, while `edk2-stable202505` is only bugged on the slower one. `master` bugs on both.

I've tried swapping the values to see if I would get `array[3] = {0x12, 0x34, 0x56, 0x00}` or similar, but that does not happen for me.

Might be related to #10258

mdkinney, 2025-05-28:
@gapalomi This is an interesting issue that needs to be root caused.

OvmfPkg.dec sets the default value to a 4 byte null-terminal ASCII string with value "yes"
```
  ## The default value UefiDriverEntryPointFwCfgOverrideLib will use in case
  #  the QEMU fw_cfg variable is not present
  gUefiOvmfPkgTokenSpaceGuid.PcdEntryPointOverrideDefaultValue|"yes"|VOID*|0x76
```

NetworkComponents.dsc.inc overrides the value for the iScsiDxe component a 3 byte null-terminal ASCII string with value "no"
```
  !if $(NETWORK_ISCSI_ENABLE) == TRUE
    NetworkPkg/IScsiDxe/IScsiDxe.inf {
      <LibraryClasses>
        UefiDriverEntryPoint|OvmfPkg/Library/UefiDriverEntryPointFwCfgOverrideLib/UefiDriverEntryPointFwCfgOverrideLib.inf
      <PcdsFixedAtBuild>
        gUefiOvmfPkgTokenSpaceGuid.PcdEntryPointOverrideFwCfgVarName|"opt/org.tianocore/ISCSISupport"
        gUefiOvmfPkgTokenSpaceGuid.PcdEntryPointOverrideDefaultValue|"no"
    }
  !endif
```

Somehow the use of of multiple processors for the build sometimes gives iScsiDxe an array length of 4 instead of 3 but always gets the correct 3 byte array with a Null-terminated ASCII string value of "no".  If single processor build is used, then the correct array length of 3 is generated.

klassiker, 2025-05-28:
> If single processor build is used, then the correct array length of 3 is generated.

To clarify: It‘s the opposite for me. Single Core results in the incorrect length.

Sometimes it also happens with multiple threads, but with decreasing chance, so I‘m guessing it has to be done on the same thread.

gapalomi, 2025-05-28:
I can take a look at it

Pconst167, 2026-09-15:
Please look at pull request for a fix: https://github.com/tianocore/edk2/pull/13124