tianocore/edk2 #11148: [Code First]: GICv5 ACPI support

jmarinho, 2025-06-02:
### Code First Item Overview

Arm has architected a new Generic Interrupt Controller (GICv5).
The GICv5 architecture definition can be found here: https://developer.arm.com/documentation/aes0070/latest/

This code-first ECR extends the MADT and SRAT tables to describe the GICv5 to OSPM.

Proposed new MADT sub-tables:
- IRS
- IWB
- ITSv5

Additionally, the ECR appends 2 new fields to the MADT GICC:
- field: IAFFID
- field: IRS ID

Proposed new SRAT sub-tables:
- IRS affinity
- IWB affinity

### What specification(s) are directly related?

ACPI

jmarinho, 2025-06-02:
[GICv5_ACPI_support_v2.docx](https://github.com/user-attachments/files/20549727/GICv5_ACPI_support_v2.docx)

mdkinney, 2025-06-10:
Thank you for the document change.  However, .docx attachments are not supported for the EDK II Code First Process.

https://github.com/tianocore/tianocore.github.io/wiki/EDK-II-Code-First-Process/

A markdown file in code first branch is where this content should be added.

jmarinho, 2025-10-17:
[0001-M2545-code-first-Add-Arm-GICv5-support_v6.patch](https://github.com/user-attachments/files/22969391/0001-M2545-code-first-Add-Arm-GICv5-support_v6.patch)

closed 2026-09-15 by mergify[bot]