tianocore/edk2 #11194: [Bug]: FMMT increasing the Binary size after replacing FFS

singhd04, 2025-06-13:
### Is there an existing issue for this?

- [x] I have searched existing issues

### Bug Type

- [x] Firmware
- [ ] Tool
- [ ] Unit Test

### What packages are impacted?

BaseTools

### Current Behavior

When the size of the incoming FFS is smaller than the existing FFS in the binary, the binary size increases unexpectedly.

### Expected Behavior

The binary image size should remain the same

### Steps To Reproduce

Run FMMT script with -r flag

"python "%batch_dir%FMMT\FMMT.py" -r"

### Build Environment

```markdown
- OS(s): Windows
- Tool Chain(s): VS2022
```

### Version Information

```text
All
```

### Urgency

High

### Are you going to fix this?

Someone else needs to fix it

### Do you need maintainer feedback?

No maintainer feedback needed

lgao4, 2025-06-19:
Seemly, this issue has been fixed at 7c6ef8eac2608e91d20c9df02930ae9b50e3fdd1. Please check.

YuweiChen1110, 2025-07-03:
After checking, it seems have a bug when padding in specific situation. Under fixing