tianocore/edk2 #11224: [Bug]: An assertion occurred: ASSERT [DxeCore] edk2/MdeModulePkg/Core/Dxe/Mem/Pool.c(730): Tail->Signature == ((('p') | ('t' << 8)) | ((('a') | ('l' << 8)) << 16))

HayatoTasaki, 2025-06-24:
### Is there an existing issue for this?

- [x] I have searched existing issues

### Bug Type

- [x] Firmware
- [ ] Tool
- [ ] Unit Test

### What packages are impacted?

MdeModulePkg

### Which targets are impacted by this bug?

RELEASE

### Current Behavior

An assertion occurred:
ASSERT [DxeCore] edk2/MdeModulePkg/Core/Dxe/Mem/Pool.c(730): Tail->Signature == ((('p') | ('t' << 8)) | ((('a') | ('l' << 8)) << 16))

The frequency of occurrence is approximately once every 1 in 5000 boots.
The hardware is a custom ARM-based board.
Please advise on how to fix this issue.

### Expected Behavior

The ASSERT [DxeCore] no longer occurs.

### Steps To Reproduce

Reboot loop triggered by sudo reboot command.

### Build Environment

```markdown
- OS(s): Ubuntu 22.04
- Tool Chain(s): aarch64--glibc--stable-2022.08-1
```

### Version Information

```text
EDK2 version:
PACKAGE_NAME = MdeModulePkg
PACKAGE_VERSION = 0.98
```

### Urgency

High

### Are you going to fix this?

Someone else needs to fix it

### Do you need maintainer feedback?

Maintainer feedback requested

HayatoTasaki, 2025-07-04:
We’d appreciate a response at your earliest convenience.

HayatoTasaki, 2025-07-07:
Thank you for your response.

I will implement a process to add hexdump logging and run the tests again.

Before reporting this issue, I also applied a fix, and I’d like to share that as additional context.

I noticed that in the CoreConnectController() function in MdeModulePkg/Core/Dxe/Hand/DriverSupport.c, when ChildHandleCount is 0, executing CoreFreePool(ChildHandleBuffer); can sometimes trigger an ASSERT [DxeCore].

To address this, I modified the logic so that when ChildHandleCount is 0, the connection of the controller handle's children is skipped entirely (i.e., no memory allocation or deallocation is performed). After applying this change, the ASSERT no longer occurs at this location.

Q1. Is this an appropriate fix?

However, I encountered another ASSERT [DxeCore] in a different location.

In the CoreConnectSingleController() function in the same file, CoreFreePool(SortedDriverBindingProtocols); occasionally triggers an ASSERT.

Before the call to CoreFreePool(SortedDriverBindingProtocols);, I logged the address of SortedDriverBindingProtocols, and also logged the buffer address received by CoreFreePoolI(). When the ASSERT occurs, the addresses differ.

Log when ASSERT occurs:
```
DriverBindingHandleCount = 43 : New = 43 : NumberOfSortedDriverBindingProtocols = 43
SortedDriverBindingProtocols = 6080847128
HeadSize = 2896 : TailSize = 2896 : TailSignature = 0 0 0 4294967296
Buffer = 6080847896
```

Log when normal:
```
DriverBindingHandleCount = 1 : New = 1 : NumberOfSortedDriverBindingProtocols = 1
SortedDriverBindingProtocols = 6089299224
HeadSize = 48 : TailSize = 48 : TailSignature = 1818326128
Buffer = 6089299224
When the ASSERT occurs, it seems that the counter used during allocation and the Head->Size calculation do not match.
```

Q2. Based on this information, do you have any ideas that could help resolve the issue?

lersek, 2025-07-07:
> I noticed that in the CoreConnectController() function in MdeModulePkg/Core/Dxe/Hand/DriverSupport.c, when ChildHandleCount is 0, executing CoreFreePool(ChildHandleBuffer); can sometimes trigger an [...]

Distinguishing `ChildHandleCount==0` should not be necessary. `MemoryAllocationLib.h` states about `AllocatePool`: *If `AllocationSize` is 0, then a valid buffer of 0 size is returned.* Consequently, releasing `ChildHandleBuffer` with `CoreFreePool` should be valid, even if the allocation size was 0 bytes. (`AllocatePool` in the `DxeCoreMemoryAllocationLib` instance simply wraps `CoreAllocatePool`, so it should not be a problem that `AllocatePool` allocates but the less abstract `CoreFreePool` releases.)

> In the CoreConnectSingleController() function in the same file, CoreFreePool(SortedDriverBindingProtocols); occasionally triggers an ASSERT.
>
> Before the call to [...]

In the normal case, you have 1 element (one pointer) in `SortedDriverBindingProtocols`. The internal allocation size (48 bytes) seems consistent with that; `SIZE_OF_POOL_HEAD` is 20 bytes, `sizeof (POOL_TAIL)` is 16 bytes, 20+1*8+16=44, which, rounded up to a multiple of 8 bytes, yields 48. The tail signature matches too (1818326128 decimal is 0x6C617470, which is `POOLPAGE_HEAD_SIGNATURE`).

In the failing case, the 2896 bytes size looks weird. If you had indeed 43 pointers in `SortedDriverBindingProtocols`, I'd expect an internal allocation size of 20+43*8+16=380; 384 after aligning up. The tail signature looks corrupted too (although I don't understand why your log contains four values there, `0 0 0 4294967296`).

The `SortedDriverBindingProtocols` pointer's value should never change between `AllocatePool` and `CoreFreePool`. In the failing case, it seems to shift by 0x300 bytes, which seems to "regular" for it to be random corruption. Still, I guess I'd sprinkle `CoreConnectSingleController` with a bunch of DEBUG messages, to narrow down where `SortedDriverBindingProtocols` changes.

HayatoTasaki, 2025-07-10:
> Distinguishing ChildHandleCount==0 should not be necessary. MemoryAllocationLib.h states about AllocatePool: If AllocationSize is 0, then a valid buffer of 0 size is returned. Consequently, [...]

To prevent the ASSERT [DxeCore], is it problematic to add logic in MdeModulePkg/Core/Dxe/Hand/DriverSupport.c that skips memory allocation when ChildHandleCount is 0?
In CoreConnectSingleController(), there is also a return statement when DriverBindingHandleCount is 0, similar to the logic we added.
What is the difference between these two implementations?

HayatoTasaki, 2025-08-06:
We added log output for investigation purposes, but the ASSERT no longer occurred. Therefore, we modified the investigation to only output a memory dump when `Tail->Signature != POOL_TAIL_SIGNATURE`.
For reference, this condition is written in the file `MdeModulePkg/Core/Dxe/Mem/Pool.c`.

After approximately 100 reboots, an ASSERT in [DxeCore] occurred twice — once during the 79th reboot and again during the 88th reboot.

Is there anything you can identify from these dumps?
We are providing the memory dumps for both occurrences below:
The memory dump is performed starting from Head and spans Head->Size bytes

[Memory dump from the 79th reboot (first ASSERT occurrence).txt](https://github.com/user-attachments/files/21609985/Memory.dump.from.the.79th.reboot.first.ASSERT.occurrence.txt)

[Memory dump from the 88th reboot (second ASSERT occurrence).txt](https://github.com/user-attachments/files/21609987/Memory.dump.from.the.88th.reboot.second.ASSERT.occurrence.txt)

- Memory dump from the 79th reboot (first ASSERT occurrence)
 00000000: 70 68 64 30 13 10 2A 01 04 00 00 00 00 00 A0 F0 
 00000010: 48 0B 00 00 00 00 00 00 98 FF 51 6C 01 00 00 00 
 00000020: 18 EF 51 6C 01 00 00 00 98 EC 51 6C 01 00 00 00 
 00000030: 18 1F 50 6C 01 00 00 00 18 1A 50 6C 01 00 00 00 
 00000040: 18 9E 4F 6C 01 00 00 00 18 14 F5 6A 01 00 00 00 
 00000050: 18 FE F3 6A 01 00 00 00 18 EF F3 6A 01 00 00 00 
 00000060: 18 E9 F3 6A 01 00 00 00 98 BF F3 6A 01 00 00 00 
 00000070: 98 B9 F3 6A 01 00 00 00 18 50 F3 6A 01 00 00 00 
 00000080: 18 59 F3 6A 01 00 00 00 18 2E F3 6A 01 00 00 00 
 00000090: 98 43 F3 6A 01 00 00 00 98 48 F3 6A 01 00 00 00 
 000000A0: 98 3B F3 6A 01 00 00 00 18 23 F3 6A 01 00 00 00 
 000000B0: 98 10 F3 6A 01 00 00 00 18 12 F3 6A 01 00 00 00 
 000000C0: 98 2C F2 6A 01 00 00 00 18 26 F2 6A 01 00 00 00 
 000000D0: 18 28 F2 6A 01 00 00 00 18 08 F3 6A 01 00 00 00 
 000000E0: 98 EE F2 6A 01 00 00 00 98 E6 F2 6A 01 00 00 00 
 000000F0: 18 E6 F2 6A 01 00 00 00 98 DE F2 6A 01 00 00 00 
 00000100: 18 9F F2 6A 01 00 00 00 18 9C F2 6A 01 00 00 00 
 00000110: 18 99 F2 6A 01 00 00 00 98 93 F2 6A 01 00 00 00 
 00000120: 98 7D F2 6A 01 00 00 00 98 78 F2 6A 01 00 00 00 
 00000130: 18 73 F2 6A 01 00 00 00 18 63 F2 6A 01 00 00 00 
 00000140: 98 5B F2 6A 01 00 00 00 98 5C F2 6A 01 00 00 00 
 00000150: 98 59 F2 6A 01 00 00 00 18 4C F2 6A 01 00 00 00 
 00000160: 98 47 F2 6A 01 00 00 00 18 3E F2 6A 01 00 00 00 
 00000170: 18 38 F2 6A 01 00 00 00 98 35 F2 6A 01 00 00 00 
 00000180: 98 15 F2 6A 01 00 00 00 98 69 F1 6A 01 00 00 00 
 00000190: 98 67 F1 6A 01 00 00 00 18 5A F1 6A 01 00 00 00 
 000001A0: 98 54 F1 6A 01 00 00 00 18 0C F2 6A 01 00 00 00 
 000001B0: 98 0E F2 6A 01 00 00 00 18 FC F1 6A 01 00 00 00 
 000001C0: 98 F2 F1 6A 01 00 00 00 18 D8 F1 6A 01 00 00 00 
 000001D0: 98 AE F1 6A 01 00 00 00 18 A7 F1 6A 01 00 00 00 
 000001E0: 98 A1 F1 6A 01 00 00 00 18 98 F1 6A 01 00 00 00 
 000001F0: 98 83 F1 6A 01 00 00 00 18 7E F1 6A 01 00 00 00 
 00000200: 98 77 F1 6A 01 00 00 00 98 72 F1 6A 01 00 00 00 
 00000210: 18 AA F0 6A 01 00 00 00 18 A4 F0 6A 01 00 00 00 
 00000220: 98 A8 F0 6A 01 00 00 00 18 4B F1 6A 01 00 00 00 
 00000230: 98 3E F1 6A 01 00 00 00 18 48 F1 6A 01 00 00 00 
 00000240: 18 39 F1 6A 01 00 00 00 18 35 F1 6A 01 00 00 00 
 00000250: 98 49 51 6C 01 00 00 00 18 48 51 6C 01 00 00 00 
 00000260: 98 39 51 6C 01 00 00 00 18 38 51 6C 01 00 00 00 
 00000270: 98 20 51 6C 01 00 00 00 18 28 51 6C 01 00 00 00 
 00000280: 18 25 51 6C 01 00 00 00 98 25 51 6C 01 00 00 00 
 00000290: 18 1C 51 6C 01 00 00 00 98 18 51 6C 01 00 00 00 
 000002A0: 98 0B 51 6C 01 00 00 00 98 0C 51 6C 01 00 00 00 
 000002B0: 18 04 51 6C 01 00 00 00 98 04 51 6C 01 00 00 00 
 000002C0: 18 FC 50 6C 01 00 00 00 18 EF 50 6C 01 00 00 00 
 000002D0: 18 EE 50 6C 01 00 00 00 98 D6 50 6C 01 00 00 00 
 000002E0: 98 C3 50 6C 01 00 00 00 98 C8 50 6C 01 00 00 00 
 000002F0: 98 BB 50 6C 01 00 00 00 98 B0 50 6C 01 00 00 00 
 00000300: 18 B7 50 6C 01 00 00 00 98 8F 50 6C 01 00 00 00 
 00000310: 18 81 50 6C 01 00 00 00 98 85 50 6C 01 00 00 00 
 00000320: 98 7E 50 6C 01 00 00 00 98 83 50 6C 01 00 00 00 
 00000330: 18 75 50 6C 01 00 00 00 18 5A 50 6C 01 00 00 00 
 00000340: 98 7E EA 6A 01 00 00 00 18 77 EA 6A 01 00 00 00 
 00000350: 18 48 F2 6A 01 00 00 00 98 6B EA 6A 01 00 00 00 
 00000360: 98 8F EE 6A 01 00 00 00 18 8D EE 6A 01 00 00 00 
 00000370: 18 81 EE 6A 01 00 00 00 18 88 EE 6A 01 00 00 00 
[... 282 lines cut ...]
 000009C0: 18 AF CD 6A 01 00 00 00 98 A1 CD 6A 01 00 00 00 
 000009D0: 98 8A CD 6A 01 00 00 00 18 7F CD 6A 01 00 00 00 
 000009E0: 98 78 CD 6A 01 00 00 00 18 DF E8 6A 01 00 00 00 
 000009F0: 18 77 CD 6A 01 00 00 00 98 D5 E8 6A 01 00 00 00 
 00000A00: 98 C0 E8 6A 01 00 00 00 18 C8 E8 6A 01 00 00 00 
 00000A10: 98 BD E8 6A 01 00 00 00 98 9D E8 6A 01 00 00 00 
 00000A20: 98 95 E8 6A 01 00 00 00 18 72 E8 6A 01 00 00 00 
 00000A30: 98 C7 E5 6A 01 00 00 00 18 AB E5 6A 01 00 00 00 
 00000A40: 18 82 E5 6A 01 00 00 00 98 0F E5 6A 01 00 00 00 
 00000A50: 18 FB E4 6A 01 00 00 00 98 F4 51 6C 01 00 00 00 
 00000A60: 98 DD E4 6A 01 00 00 00 18 DA E4 6A 01 00 00 00 
 00000A70: 18 C0 E4 6A 01 00 00 00 98 AE E4 6A 01 00 00 00 
 00000A80: 98 9D E4 6A 01 00 00 00 18 5F E4 6A 01 00 00 00 
 00000A90: 98 6D E4 6A 01 00 00 00 98 7E E4 6A 01 00 00 00 
 00000AA0: 98 5D E4 6A 01 00 00 00 18 6C E4 6A 01 00 00 00 
 00000AB0: 18 52 E4 6A 01 00 00 00 98 AC E3 6A 01 00 00 00 
 00000AC0: 18 AE E2 6A 01 00 00 00 18 A3 E2 6A 01 00 00 00 
 00000AD0: 18 A9 E2 6A 01 00 00 00 98 9C E2 6A 01 00 00 00 
 00000AE0: 98 99 E2 6A 01 00 00 00 98 8B E2 6A 01 00 00 00 
 00000AF0: 98 87 E2 6A 01 00 00 00 98 71 E2 6A 01 00 00 00 
 00000B00: 98 78 E2 6A 01 00 00 00 18 6A E2 6A 01 00 00 00 
 00000B10: 98 66 E2 6A 01 00 00 00 98 5F E2 6A 01 00 00 00 
 00000B20: 98 58 E2 6A 01 00 00 00 98 4B E2 6A 01 00 00 00 
 00000B30: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
 00000B40: 48 0B 00 00 00 00 00 00

lersek, 2025-08-06:
1. When I [previously wrote](https://github.com/tianocore/edk2/issues/11224#issuecomment-3046628561) that `SIZE_OF_POOL_HEAD` was 20 bytes, I was wrong. I failed to take field alignment into account. The size is 24 bytes in reality. On AARCH64, there is 4 bytes of padding just before `POOL_HEAD.Size`, as the latter is of type `UINTN` (i.e., `UINT64`), and gets aligned on an 8-byte boundary. The calculation I wrote in that comment remains correct, though, ultimately, it's just that the in-struct padding obviates the allocation round-up (**24**+1*8+16=48, **24**+43*8+16=384).
2. Each of both dumps seems to consist of: (a) a `POOL_HEAD` (24 bytes), (b) 356 pointers in an array (2848 bytes), (c) a partially overwritten `POOL_TAIL` (16 bytes). Total: 2888 bytes.
3. In `POOL_HEAD`, `Signature` is `phd0` (`70 68 64 30`), `Reserved` is garbage (4 bytes), `Type` is `EfiBootServicesData` (04 00 00 00), then we have four bytes of padding (garbage); finally, `Size` (of type UINTN) is `0xb48` (`48 0b 00 00 00 00 00 00`).
4. The 8-byte pointers in the 356-element array are represented in little endian. Converting this representation to actual values, we get pointers such as `0x000000016c51ff98`. The pointer array is not sorted in either ascending or descending order.
5. The two dumps differ in very small details from each other: (a) the `Reserved` field is filled with (different) garbage in each, (b) so are the 4 padding bytes just before the `Size` field, (c) the pointer entries at absolute file offsets 0xA78 and 0xA80 have single-byte differences (one byte per entry) between both dumps.
6. The interesting bit is the (partially overwritten) `POOL_TAIL`. This is the end of the hexdump (in either dump):

~~~
00000b20  98 58 e2 6a 01 00 00 00  98 4b e2 6a 01 00 00 00  |.X.j.....K.j....|
00000b30  00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00  |................|
00000b40  48 0b 00 00 00 00 00 00                           |H.......|
00000b48
~~~

Observe the UINT64 value at offset `0xb40`: it's the LE representation of value `0xb48`: this is the `POOL_TAIL.Size` field.

The `POOL_TAIL.Signature` and `POOL_TAIL.Reserved` fields start at offset `0xb38` -- but they've been zeroed out! (Well, zero is a fine value for `Reserved`, but not for `Signature`.)

What's more, we have a pointer element (!) with value `0x0` at offset 0xb30. It doesn't look like a valid pointer to `EFI_DRIVER_BINDING_PROTOCOL`.

You mentioned `SortedDriverBindingProtocols` before. I now propose that the logic that populates this array, with pointers to `EFI_DRIVER_BINDING_PROTOCOL`, is buggy. First, it seems to carry a useless (or otherwise inexplicable) null pointer at the end of the array. Second, it even overflows the allocated buffer by one pointer element -- which too happens to be a null pointer. Said overflow corrupts the `POOL_TAIL` signature.

I suggest working backwards from here. Try to catch calls when room is allocated for 356 pointers, and when 356 pointers are placed into the array. Try to add debug logs to the entry-addition logic; see what it does to the buffer step by step. I think there may be a corner case in the logic that fills in `SortedDriverBindingProtocols` that runs off the end of the buffer.

lersek, 2025-08-06:
You could also try enabling the heap guard; that way, the buffer overflow should crash immediately.

lersek, 2025-08-07:
> Could you please explain the basic steps to enable heap guard in EDK II?

Refer to `PcdHeapGuardPropertyMask` and `PcdHeapGuardPoolType` in `MdeModulePkg/MdeModulePkg.dec`.

In your platform DSC, I believe that you should set those PCDs as follows:
- `PcdHeapGuardPropertyMask|2`. Explanation: enable the UEFI pool goard (BIT1 set), and make sure that the returned buffer pointer is near the tail guard page (BIT7 clear). The symptom is that we run off the tail end of the allocated block, so this "anchor direction" should ensure that the buffer overflow triggers a page fault immediately.
- `PcdHeapGuardPoolType|0x10`. Explanation: from the hexdump, we know that it's an `EfiBootServicesData` allocation that is overflowed.

> I’ll start investigating where memory is allocated for 356 pointers and how they are added to the array, and I’ll add debug logs to observe the buffer behavior step by step.

You can check for the allocation size right in the memory allocator, and call `CpuDeadLoop` or `CpuBreakpoint` for call stack investigation (or at least for identifying the caller module).

lersek, 2025-08-08:
@HayatoTasaki 

> After enabling Heap Guard, I consistently encounter the following assertion during boot. This makes it difficult to investigate the original assertion I was targeting.
> 
> `Synchronous Exception at 0x000000013B1850B8 ASSERT [ArmCpuDxe] /work/edk2/ArmPkg/Library/DefaultExceptionHandlerLib/AArch64/DefaultExceptionHandler.c(345): ((BOOLEAN)(0==1))`

This seems to me like the inteded effect. The pool guard may be catching the buffer overflow for you, every time.

I'm uncertain what release of edk2 you are using; the line number `345` is a bit off from what's in `DefaultExceptionHandler.c`, as of commit b58ce4c22676. The `DefaultExceptionHandler` function ends like this:
~~~
    318   DEBUG ((DEBUG_ERROR, "\nStack dump:\n"));
    319   for (Offset = -256; Offset < 256; Offset += 32) {
    320     DEBUG ((
    321       DEBUG_ERROR,
    322       "%c %013lx: %016lx %016lx %016lx %016lx\n",
    323       Offset == 0 ? '>' : ' ',
    324       SystemContext.SystemContextAArch64->SP + Offset,
    325       *(UINT64 *)(SystemContext.SystemContextAArch64->SP + Offset),
    326       *(UINT64 *)(SystemContext.SystemContextAArch64->SP + Offset + 8),
    327       *(UINT64 *)(SystemContext.SystemContextAArch64->SP + Offset + 16),
    328       *(UINT64 *)(SystemContext.SystemContextAArch64->SP + Offset + 24)
    329       ));
    330   }
    331 
    332   // Attempt to print that we had a synchronous exception to ConOut.  We do
    333   // this after the serial logging as ConOut's logging is more complex and we
    334   // aren't guaranteed to succeed.
    335   if (gST->ConOut != NULL) {
    336     gST->ConOut->OutputString (gST->ConOut, UnicodeBuffer);
    337   }
    338 
    339   ASSERT (FALSE);
    340   CpuDeadLoop ();
~~~

Meaning that **between** the `Synchronous Exception` message and the `ASSERT`, you should be seeing a register dump and a stack dump.

> The method you suggested didn’t change the values of gEfiMdeModulePkgTokenSpaceGuid.PcdHeapGuardPropertyMask and gEfiMdeModulePkgTokenSpaceGuid.PcdHeapGuardPoolType from 0. Therefore, I enabled them by adding the following lines to MdeModulePkg/MdeModulePkg.dsc
> 
> `[PcdsFixedAtBuild] gEfiMdeModulePkgTokenSpaceGuid.PcdHeapGuardPropertyMask|0x2 gEfiMdeModulePkgTokenSpaceGuid.PcdHeapGuardPoolType|0x10 `

Sorry, this makes no sense to me. I wrote (emphasis added here):

> In your **platform DSC**, I believe that you should set those PCDs as follows:

Changing PCD values in `MdeModulePkg.dsc` only makes a difference if you invoke the `build` utility with `-p MdeModulePkg/MdeModulePkg.dsc`. You must surely have a platform DSC file for your own platform; for example, if you wanted to build the ArmVirtQemu platform, you'd modify `ArmVirtPkg/ArmVirtQemu.dsc` and invoke the `build` utility with `-p ArmVirtPkg/ArmVirtQemu.dsc`. The `build` utility would then compose a firmware image according to `ArmVirtPkg/ArmVirtQemu.fdf`.

Building `MdeModulePkg.dsc` gives you a number of standalone binary files (= library files + executable modules (PEIMs, DXE drivers, UEFI drivers, UEFI applications)), but no firmware image (because there is no FDF file referenced in the DSC file, with a `FLASH_DEFINITION` entry). I'm unsure why you'd want that -- but certainly, if that *is* what you want (for wharever reason), then `MdeModulePkg.dsc` *is* your platform DSC file, and what you ended up doing actually matches what I advised.

HayatoTasaki, 2025-09-03:
It is becoming clear that memory corruption in `SortedDriverBindingProtocols` located in `MdeModulePkg/Core/Dxe/Hand/DriverSupport.c` is not the cause of the `ASSERT[DxeCore]`.

I checked whether the Tail-Signature was corrupted immediately before every call to `CoreFreePool()` in files under `MdeModulePkg/Core/Dxe/`. When the `ASSERT[DxeCore]` occurred, the Tail-Signature was intact just before `CoreFreePool()` was called.  
Additionally, I confirmed that the `ASSERT[DxeCore]` can occur after returning from `CoreFreePool()` when the Tail-Signature was not corrupted at the time of the call.

I have two questions:

1. Is it likely that the `ASSERT[DxeCore]` is triggered inside `CoreFreePool()` when called from `FreePool()` in `MdeModulePkg/Library/DxeCoreMemoryAllocationLib/MemoryAllocationLib.c`?

2. Can we trace the caller of `FreePool()` in case of memory corruption? 
    For example, is it possible to obtain a backtrace?

HayatoTasaki, 2025-09-09:
@lersek Thank you so much for your help with this issue. I really appreciate your support!

HayatoTasaki, 2025-09-09:
> It is becoming clear that memory corruption in `SortedDriverBindingProtocols` located in `MdeModulePkg/Core/Dxe/Hand/DriverSupport.c` is not the cause of the `ASSERT[DxeCore]`.
>
> I checked whether [...]

It’s becoming clear that this is not the root cause. I had overlooked the fact that the function pointer mBootServices->CoreFreePool exists in MdeModulePkg/Core/Dxe/DxeMain/DxeMain.c, and I’ve found that an ASSERT[DxeCore] is triggered during this call. Is there an easy way to identify which function is calling mBootServices->CoreFreePool at the time of memory corruption? Is it possible to get a backtrace?