tianocore/edk2 #11281: Microsoft UEFI CA 2023

Supericy, 2025-07-10:
Looks like EDK2 enrolls Microsoft's 2011 CA by default (https://github.com/tianocore/edk2/blob/master/OvmfPkg/EnrollDefaultKeys/AuthData.c). Are there any plans to incorporate `Microsoft UEFI CA 2023` into the default set?

https://github.com/microsoft/secureboot_objects

Flickdm, 2025-07-21:
FYI - @kraxel, @ardbiesheuvel, @jyao1

Supericy, 2025-07-22:
Thank you for the responses!

I have a local patch that I've been using that enrolls these 3 keys:
- https://github.com/microsoft/secureboot_objects/blob/main/PreSignedObjects/DB/Certificates/microsoft%20uefi%20ca%202023.der
- https://github.com/microsoft/secureboot_objects/blob/main/PreSignedObjects/DB/Certificates/windows%20uefi%20ca%202023.der
- https://github.com/microsoft/secureboot_objects/blob/main/PreSignedObjects/KEK/Certificates/microsoft%20corporation%20kek%202k%20ca%202023.der

Works nicely for my use case. I'm happy to make a PR. In your mind, do these 3 keys make sense to add? Should any others be added?

lersek, 2025-07-24:
(A certificate for verifying drivers in PCIe expansion ROMs could be useful with GPU or other device assignment.)

alexhaydock, 2025-10-26:
> virt-firmware (https://gitlab.com/kraxel/virt-firmware) offers an easier and more flexible way to enroll secure boot certificates into ovmf variable stores, so updating `EnrollDefaultKeys.efi` hasn't been a priority.
> 
> PRs are welcome though.

@kraxel I'm interested in possibly authoring a PR to fix this one, but just wanted to check that I'm right in thinking that `EnrolDefaultKeys.efi` does form part of the chain that builds the default Secure Boot VARS files for downstream packages?

Using the `virt-fw-vars` tool to list the keys enrolled in `OVMF_VARS_4M.ms.fd` from Debian, it is clear that they're only shipping the Microsoft 2011 key. So that would make sense.

But when using it on Fedora's `OVMF_VARS_4M.secboot.qcow2` equivalent, the Microsoft 2023 key is also being included. So it seems Fedora has a different method for pre-enrolling the `OVMF_VARS_*` files. I can't quite work out where that's being introduced though.

kraxel, 2025-10-27:
> [@kraxel](https://github.com/kraxel) I'm interested in possibly authoring a PR to fix this one, but just wanted to check that I'm right in thinking that `EnrolDefaultKeys.efi` does form part of the chain that builds the default Secure Boot VARS files for downstream packages?

It used to be the only option to enroll keys.

> Using the `virt-fw-vars` tool to list the keys enrolled in `OVMF_VARS_4M.ms.fd` from Debian, it is clear that they're only shipping the Microsoft 2011 key. So that would make sense.

Yes, looks like debian still uses `EnrolDefaultKeys`.

> But when using it on Fedora's `OVMF_VARS_4M.secboot.qcow2` equivalent, the Microsoft 2023 key is also being included. So it seems Fedora has a different method for pre-enrolling the `OVMF_VARS_*` files. I can't quite work out where that's being introduced though.

Fedora switched to `virt-fw-vars` in 2022.