tianocore/edk2 #11360: [Bug]: TCG2/TPM2 support on RISC-V does not work

Vogtinator, 2025-07-30:
### Is there an existing issue for this?

- [x] I have searched existing issues

### Bug Type

- [x] Firmware
- [ ] Tool
- [ ] Unit Test

### What packages are impacted?

OvmfPkg

### Which targets are impacted by this bug?

DEBUG

### Current Behavior

When booting OVMF in QEMU, no Tcg2Dxe is registered and many instances of

```
None of Tcg2Protocol/CcMeasurementProtocol is installed.
```

are visible. The OS sees the TPM2 device though but it fails selfcheck.

### Expected Behavior

Like on aarch64 and x86_64, the TPM should be detected and measurements performed.

### Steps To Reproduce

Build OVMF using

```
build -D TPM2_ENABLE -D TPM2_CONFIG_ENABLE -a RISCV64 -p OvmfPkg/RiscVVirt/RiscVVirtQemu.dsc -b DEBUG -t GCC5
truncate -s 32M Build/RiscVVirtQemu/DEBUG_GCC5/FV/RISCV_VIRT_CODE.fd
```

Start `swtpm` using

```
mkdir /tmp/tpm0
swtpm socket --tpmstate dir=/tmp/tpm0 --ctrl type=unixio,path=/tmp/tpm0/swtpm-sock --tpm2 --log level=20
```

and start emulation using

```
qemu-system-riscv64 -M virt -serial stdio -drive file=RISCV_VIRT_CODE.fd,if=pflash,readonly=on -chardev socket,id=chrtpm,path=/tmp/tpm0/swtpm-sock -tpmdev emulator,id=tpm0,chardev=chrtpm -device tpm-tis-device,tpmdev=tpm0
```

### Build Environment

```markdown
- OS(s): openSUSE Tumbleweed
- Tool Chain(s): GCC5
```

### Version Information

```text
Commit: a60334ad59eb8b6f2f056febeb0c2b5ed20c7b57
```

### Urgency

Medium

### Are you going to fix this?

Someone else needs to fix it

### Do you need maintainer feedback?

Maintainer feedback requested

### Anything else?

FWICT the needed DXE drivers are not part of the FV.

Just adding them was not enough though:

```diff
diff --git a/OvmfPkg/RiscVVirt/RiscVVirtQemu.fdf b/OvmfPkg/RiscVVirt/RiscVVirtQemu.fdf
index 3470b18a2e..12e1e6863e 100644
--- a/OvmfPkg/RiscVVirt/RiscVVirtQemu.fdf
+++ b/OvmfPkg/RiscVVirt/RiscVVirtQemu.fdf
@@ -211,6 +211,16 @@ INF  MdeModulePkg/Bus/Usb/UsbBusDxe/UsbBusDxe.inf
 INF  MdeModulePkg/Bus/Usb/UsbKbDxe/UsbKbDxe.inf
 INF  MdeModulePkg/Bus/Usb/UsbMassStorageDxe/UsbMassStorageDxe.inf
 
+#^M
+# TPM2 support^M
+#^M
+!if $(TPM2_ENABLE) == TRUE^M
+INF SecurityPkg/Tcg/Tcg2Dxe/Tcg2Dxe.inf^M
+!if $(TPM2_CONFIG_ENABLE) == TRUE^M
+INF SecurityPkg/Tcg/Tcg2Config/Tcg2ConfigDxe.inf^M
+!endif^M
+!endif^M
+^M
 #
 # TianoCore logo (splash screen)
 #

```

lgao4, 2025-08-07:
@Vogtinator , please investigate and add this support.

Vogtinator, 2025-08-07:
See

> Are you going to fix this?
>
> Someone else needs to fix it

I'm not familiar enough with OVMF/EDK2 to do this contribution.

lgao4, 2025-08-20:
> [@lgao4](https://github.com/lgao4) Please unassign me, someone else will have to do this.

Done.

Vogtinator, 2025-11-27:
Oh, and I think support for the "TPM2" ACPI table generation is also missing, so I had to use qemu with `-M virt,acpi=off` to make the TPM visible to Linux.

pttuan, 2025-11-27:
@Vogtinator: Looks like I missed that code for PEI in FDF during merging. Will fix it.
About ACPI table, QEMU hardware target for RISC-V is missing code to generate TPM2 device inside DSDT. I will send a patch for QEMU to fix it.

Vogtinator, 2026-01-07:
With #11829 merged, I think this can be closed now. Thanks!

closed 2026-01-07 by Vogtinator