tianocore/edk2 #11363: [Bug]: Hit LNK1237 when using IntrinsicLib.inf

JonsonTsai1, 2025-07-31:
### Is there an existing issue for this?

- [x] I have searched existing issues

### Bug Type

- [x] Firmware
- [ ] Tool
- [ ] Unit Test

### What packages are impacted?

MdePkg, CryptoPkg

### Which targets are impacted by this bug?

DEBUG, RELEASE

### Current Behavior

My DXE driver will hit LNK1237 error when using IntrinsicLib.

```
BaseIntrinsicLib.lib(MemoryIntrinsics.obj) : fatal error LNK1237: during code generation, compiler introduced reference to symbol 'CompareMem' defined in module 'BaseMemoryLibRepStr.lib(CompareMemWrapper.obj)' compiled with /GL

Finished generating code
NMAKE : fatal error U1077: 'C:\BEA\BuildTools\VS2019_5d18eea\VC\bin\Hostx86\x64\link.exe' : return code '0x4d5'
Stop.

```

The problem is raised because IntrinsicLib.inf overwrites CC_FLAGS to remove /GL while it references to BaseMemoryLib which is built with /GL flag.

https://learn.microsoft.com/en-us/cpp/error-messages/tool-errors/linker-tools-error-lnk1237?view=msvc-170

I've checking why /GL f lag is removed in the log. However it gets me further confused: https://github.com/tianocore/edk2/commit/f754e613fcaff79255104d82f8a5a01dc4c50d49

It said the modification is to:
1. enable /GL optimization building on OpensslLib.
2. add DDK3790 compiler option for IA32.

However the code is inconsistent with the change in IntrinsicLib.inf

### Expected Behavior

Build process should be completed without stop.

### Steps To Reproduce

In your DXE driver, C code, add the following lines:

static UINT8 array[20] = { 0 }

This will introduce memset() call by C compiler.

In the INF file, add IntrinsicLib into [LibraryClasses] 

build your module....then LNK1237 error hit.

### Build Environment

```markdown
- OS(s): Windows 11
- Tool Chain(s): VS2019
```

### Version Information

```text
Commit: EDK2 later than f754e61
```

### Urgency

Medium

### Are you going to fix this?

Someone else needs to fix it

### Do you need maintainer feedback?

Maintainer feedback requested

lgao4, 2025-08-07:
@JonsonTsai1 , could you give the full log and code base version?

@liyi77 , could you help check this build failure about Openssl?

JonsonTsai1, 2025-08-07:
[EDK2_Build_CryptoPkg.txt](https://github.com/user-attachments/files/21661018/EDK2_Build_CryptoPkg.txt) 

This is the build log in Windows environment. 
Steps: 

1. Open a command prompt at EDK2 folder
2. Modify Config\Target.txt
3. Modify ACTIVE_PLATFORM = CryptoPkg/CryptoPkg.dsc 
4. Run Edk2Setup.bat Rebuild
5. Run Edk2Setup.bat
6. Run set PYTHON_HOME=C:\Program Files\Python310
7. Run set NASM_PREFIX=C:\BEA\AgS_2024\DellPkgs\BuildTools\NASM\
8. Run set IASL_PREFIX=C:\BEA\AgS_2024\DellPkgs\BuildTools\ASL\
9. You could find LNK1237 error