tianocore/edk2 #11374: [Bug]: UEFI runtime drivers loaded after EndOfDxe are not executable by OS

vit9696, 2025-08-02:
### Is there an existing issue for this?

- [x] I have searched existing issues

### Bug Type

- [x] Firmware
- [ ] Tool
- [ ] Unit Test

### What packages are impacted?

MdeModulePkg, OvmfPkg

### Which targets are impacted by this bug?

DEBUG, NOOPT, RELEASE

### Current Behavior

We load a runtime Driver after EndOfDxe event and then call it from the operating system (OS).

As per [this source code reference](https://github.com/tianocore/edk2/blob/964c22b/MdeModulePkg/Core/Dxe/Misc/MemoryAttributesTable.c#L594-L597) UEFI runtime Drivers loaded after EndOfDxe are not added to `MemoryAttribytesTable`.

* In the past this meant they would get RWX permissions in the OS in some firmware, this was reported as #9906.
* Now https://github.com/tianocore/edk2/commit/bb248a95091ab542440053d9c289a97e80eb6630 introduced a regression, making these drivers get R/O permissions.

As a result of this change (https://github.com/tianocore/edk2/commit/bb248a95091ab542440053d9c289a97e80eb6630) attempting to execute runtime Driver (loaded after EndOfDxe) code from the OS results in a page fault.

### Expected Behavior

As per UEFI specification it is allowed for the UEFI application to load the runtime drivers and they should be available after a UEFI OS loader has taken control ([specification reference](https://uefi.org/specs/UEFI/2.11/02_Overview.html#uefi-drivers)).

### Steps To Reproduce

Our use case is a bit more complicated, but it is possible to reproduce the issue the following way:
 
1. Take OVMF (x64)
2. Load a simple runtime driver that will override any Runtime Service, e.g. SetVariable.
3. Try to load Linux, e.g. Debian 12 installer.

The following crash will be observed with console=ttyS0:

```
[    0.757347] kernel tried to execute NX-protected page - exploit attempt? (uid: 0)
[    0.758306] BUG: unable to handle page fault for address: fffffffeff7f5faa
[    0.758607] #PF: supervisor instruction fetch in kernel mode
[    0.758764] #PF: error_code(0x0011) - permissions violation
[    0.759045] PGD 50704067 P4D 50704067 PUD 50700063 PMD 11c4063 PTE 800000007f5f5061
[    0.759884] Oops: 0011 [#1] PREEMPT SMP PTI
[    0.759946] CPU: 0 PID: 0 Comm: swapper/0 Not tainted 6.1.0-35-amd64 #1  Debian 6.1.137-1
[    0.759946] Hardware name: Acidanthera Macmini8,1/Mac-7BA5B2DFE22DDD8C, BIOS 2069.0.0.0.0 08/07/2024
[    0.759946] RIP: 0010:0xfffffffeff7f5faa
[    0.759946] Code: 00 5b 5f 5e 41 5c 41 5d 41 5e 41 5f 5d c3 66 83 a5 c0 fd ff ff 00 48 8d 8d b0 fd ff ff 6a 10 5a e8 58 f1 ff ff e9 72 fd ff ff <55> 48 89 e5 41 57 41 56 41 55 41 54 56 57 53 48 81 ec 48 02 00 00
[    0.759946] RSP: 0000:ffffffffbce03e08 EFLAGS: 00010282
[    0.759946] RAX: 0000000000000202 RBX: 8000000000000006 RCX: ffffffffbc40a2b8
[    0.759946] RDX: ffffffffbce03e90 RSI: ffffffffbc40a2b8 RDI: fffffffeff7f5faa
[    0.759946] RBP: ffffffffbce03e40 R08: 0000000000000007 R09: 0000000000000000
[    0.759946] R10: 0000000000000048 R11: ffffffffbced4400 R12: ffffffffbce03e90
[    0.759946] R13: 0000000000000007 R14: 0000000000000000 R15: 0000000000000000
[    0.759946] FS:  0000000000000000(0000) GS:ffff9cc0bb600000(0000) knlGS:0000000000000000
[    0.759946] CS:  0010 DS: 0000 ES: 0000 CR0: 0000000080050033
[    0.759946] CR2: fffffffeff7f5faa CR3: 00000000011ca000 CR4: 00000000001506f0
[    0.759946] Call Trace:
[    0.759946]  <TASK>
[    0.759946]  ? __efi_call+0x28/0x30
[    0.759946]  virt_efi_set_variable_nonblocking+0xb0/0x120
[    0.759946]  efi_delete_dummy_variable+0x5c/0x80
[    0.759946]  efi_enter_virtual_mode+0x429/0x44b
[    0.759946]  start_kernel+0x6a1/0x733
[    0.759946]  secondary_startup_64_no_verify+0xe5/0xeb
[    0.759946]  </TASK>
[    0.759946] Modules linked in:
[    0.759946] CR2: fffffffeff7f5faa
[    0.759946] ---[ end trace 0000000000000000 ]---
[    0.759946] RIP: 0010:0xfffffffeff7f5faa
[    0.759946] Code: 00 5b 5f 5e 41 5c 41 5d 41 5e 41 5f 5d c3 66 83 a5 c0 fd ff ff 00 48 8d 8d b0 fd ff ff 6a 10 5a e8 58 f1 ff ff e9 72 fd ff ff <55> 48 89 e5 41 57 41 56 41 55 41 54 56 57 53 48 81 ec 48 02 00 00
[    0.759946] RSP: 0000:ffffffffbce03e08 EFLAGS: 00010282
[    0.759946] RAX: 0000000000000202 RBX: 8000000000000006 RCX: ffffffffbc40a2b8
[    0.759946] RDX: ffffffffbce03e90 RSI: ffffffffbc40a2b8 RDI: fffffffeff7f5faa
[    0.759946] RBP: ffffffffbce03e40 R08: 0000000000000007 R09: 0000000000000000
[    0.759946] R10: 0000000000000048 R11: ffffffffbced4400 R12: ffffffffbce03e90
[    0.759946] R13: 0000000000000007 R14: 0000000000000000 R15: 0000000000000000
[    0.759946] FS:  0000000000000000(0000) GS:ffff9cc0bb600000(0000) knlGS:0000000000000000
[    0.759946] CS:  0010 DS: 0000 ES: 0000 CR0: 0000000080050033
[    0.759946] CR2: fffffffeff7f5faa CR3: 00000000011ca000 CR4: 00000000001506f0
[    0.759946] Kernel panic - not syncing: Attempted to kill the idle task!
[    0.759946] ---[ end Kernel panic - not syncing: Attempted to kill the idle task! ]—-
```

### Build Environment

```markdown
- OS(s): macOS 15, but also reproducible on Ubuntu 24.04.
- Tool Chain(s): XCODE5, but also reproducible with GCC
```

### Version Information

```text
edk2/master (7fe3609022ce7f52190bd6384d117f1510e15afe)

Bug introduced in https://github.com/tianocore/edk2/commit/bb248a95091ab542440053d9c289a97e80eb6630.
```

### Urgency

Medium

### Are you going to fix this?

I will fix it

### Do you need maintainer feedback?

Maintainer feedback requested

### Anything else?

I suggest to remove [MemoryAttributesTable.c#L594-L597 ](https://github.com/tianocore/edk2/blob/964c22b/MdeModulePkg/Core/Dxe/Misc/MemoryAttributesTable.c#L594-L597) lines to fix the issue. Can submit a patch if there are no objections.

vit9696, 2025-08-02:
CC @os-d, @ardbiesheuvel as the ones in charge for the offending changes.

os-d, 2025-08-04:
@ardbiesheuvel what are your thoughts here? Looks like the end of DXE restriction came from the old properties table: https://github.com/tianocore/edk2/commit/4a723ed258367471eac8b4ae32558f09ef65672e.

I think that we should have all loaded runtime drivers described in the MAT. Some platforms will take a security posture based on end of DXE and I think it would be reasonable to say no drivers can be loaded after end of DXE, but if edk2 wants to take that stance, I think it should be in the loader, not here. i.e. we should block images from getting loaded if we don't want runtime images loaded then, not load them and then not report them.

There are definitely issues that could arise from when an image is loaded and added to the MAT and when the bootloader consumes the MAT which make any loading of images after end of DXE dangerous.