tianocore/edk2 #11394: [Bug]: BSP could free AP reset vector buffer even though AP wake up is in process

philnoh2, 2025-08-06:
### Is there an existing issue for this?

- [x] I have searched existing issues

### Bug Type

- [x] Firmware
- [ ] Tool
- [ ] Unit Test

### What packages are impacted?

UefiCpuPkg

### Which targets are impacted by this bug?

DEBUG, RELEASE

### Current Behavior

If AP loop mode is not Halt loop, it is found that the AP in ApInitDone state exits the loop mode immediately when WAKEUP_AP_SIGNAL is signaled at the first step of AP wake up. When WakeUpByInitSipiSipi is enabled, it is possible that the AP could clear StartupApSignal buffer before the BSP sends Init-Startup IPI to the AP. In the case, it causes that the BSP frees AP reset vector buffer even though AP wake up is in process. This results in an exception error.

For example, the related code in WakeUpAp function is as follow.
```c
WakeUpAP ()
{
  ...
  } else {
    ...
    //
    // Wakeup specified AP
    //
    ASSERT (CpuMpData->InitFlag == ApInitDone);
    *(UINT32 *)CpuData->StartupApSignal = WAKEUP_AP_SIGNAL;
    if (ResetVectorRequired) {
      ...
      } else {
        SendInitSipiSipi (
          CpuInfoInHob[ProcessorNumber].ApicId,
          (UINT32)ExchangeInfo->BufferStart
          );
      }
    }

    //
    // Wait specified AP waken up
    //
    WaitApWakeup (CpuData->StartupApSignal);
  }

  if (ResetVectorRequired) {
    FreeResetVector (CpuMpData);
  }
  ...
}
``` 

### Expected Behavior

 To ensure AP wake up, it is expected that the AP is in the known non-executable state before the AP wakeup signal.

### Steps To Reproduce

1. Boot to EFI shell
2. Run an MP test tool that enables and disables all APs sequentially through the following service
    gEfiMpServiceProtocolGuid->EnableDisableAP service
3. The error was detected by the stress test (e.g. 10000 times) in the tool

### Build Environment

```markdown
- OS(s): EFI Shell
- Tool Chain(s): VS2022
```

### Version Information

```text
Tag: edk2-stable202505
```

### Urgency

Medium

### Are you going to fix this?

I will fix it

### Do you need maintainer feedback?

Maintainer feedback requested

### Anything else?

I have committed the following PR that fixes the issue. I have got some feedback from EDK2 maintainers.
At this point, I am requesting EDK2 maintainers to review the latest version on the feedback.

https://github.com/tianocore/edk2/pull/11215

lgao4, 2025-08-07:
PR has been submitted.

philnoh2, 2025-09-05:
The PR (https://github.com/tianocore/edk2/pull/11215) for this issue has been approved and merged. Close the issue report.

closed 2025-09-05 by philnoh2