tianocore/edk2 #11428: [Bug]: Window layout issue in Boot Manager when running qemu-system-x86_64 with -nographic

r1chard-lyu, 2025-08-19:
### Is there an existing issue for this?

- [x] I have searched existing issues

### Bug Type

- [x] Firmware
- [ ] Tool
- [ ] Unit Test

### What packages are impacted?

OvmfPkg

### Current Behavior

After commit `9224a2b917`, the Boot Manager shows a window layout issue when running under qemu-system-x86_64 with the `-nographic` option. The issue only occurs in `-nographic` option, while the VNC window does not have this issue.

### Expected Behavior

The Boot Manager should also display normally when using `-nographic`, just like in the VNC window.

### Steps To Reproduce

1. git checkout 9224a2b917
2. build -a X64 -p OvmfPkg/OvmfPkgX64.dsc -t GCC5
3. qemu-system-x86_64 -pflash OVMF.fd -nographic

### Build Environment

```markdown
- OS(s): openSUSE Leap 16.0
- Tool Chain(s): GCC5
```

### Version Information

```text
Commit: 9224a2b917
```

### Urgency

Medium

### Are you going to fix this?

I will fix it

### Do you need maintainer feedback?

No maintainer feedback needed

EricGao2015, 2025-08-25:
It seems PR https://github.com/tianocore/edk2/pull/11295 could solve this issue, you can give a try.

r1chard-lyu, 2025-08-25:
I tested PR #11295, and it resolved the terminal layout issue at {160, 42}.

closed 2025-09-01 by r1chard-lyu