tianocore/edk2 #11453: [Bug]: AsciiSPrint does not handle %s correctly

ddaskala, 2025-08-29:
### Is there an existing issue for this?

- [x] I have searched existing issues

### Bug Type

- [x] Firmware
- [ ] Tool
- [ ] Unit Test

### What packages are impacted?

OvmfPkg

### Which targets are impacted by this bug?

DEBUG, RELEASE

### Current Behavior

AsciiSPrint has a bug where it does not handle format strings with %s properly. I haven't been able to track it down in the EDK2 source, but running the following code snippet will show the bad behavior.

I suspect it is related to using the wrong type when converting VaLists to BaseLists in DxePrintLibPrint2ProtocolVaListToBaseList.

```
{
    CHAR8 Ascii[20];
    CHAR16 Unicode[20];
    AsciiSPrint(Ascii, 20, "%s", "ascii string");
    AsciiStrToUnicodeStrS(Ascii, Unicode, 20);
    DEBUG((DEBUG_ERROR, "Unicode String: %s\n", Unicode));
}
```

Output:
Unicode String: acisrn

### Expected Behavior

```
{
    CHAR8 Ascii[20];
    CHAR16 Unicode[20];
    AsciiSPrint(Ascii, 20, "%s", "ascii string");
    AsciiStrToUnicodeStrS(Ascii, Unicode, 20);
    DEBUG((DEBUG_ERROR, "Unicode String: %s\n", Unicode));
}
```

Output:
Unicode String: ascii string

### Steps To Reproduce

```
{
    CHAR8 Ascii[20];
    CHAR16 Unicode[20];
    AsciiSPrint(Ascii, 20, "%s", "ascii string");
    AsciiStrToUnicodeStrS(Ascii, Unicode, 20);
    DEBUG((DEBUG_ERROR, "Unicode String: %s\n", Unicode));
}
```

### Build Environment

```markdown
- OS(s): CentOS Stream 9 running Linux 6.9
- Tool Chain(s): edk2-ovmf version 4.14.0, GCC5
```

### Version Information

```text
Commit: d46aa46c8361194521391aa581593e556c707c6e
```

### Urgency

Low

### Are you going to fix this?

Someone else needs to fix it

### Do you need maintainer feedback?

No maintainer feedback needed

mdkinney, 2025-08-29:
%s means a Unicode string, but an ASCII string is being passed in.  %a means an ASCII string.

Either change %s to %a or change "ascii string" to L"ascii string".

ddaskala, 2025-08-29:
I see. In our case we have a compatiblity layer to reuse some code from a Linux driver within the UEFI driver. So the format string follows the conventions of the standard C library.

I assume this is because the UEFI string formatting is shared between unicode/ascii helpers, so it won't be easy to support the standard %s?

mdkinney, 2025-08-29:
The PrintLib in edk2 does differ from standard C lib for some format string semantics.  

https://github.com/tianocore/edk2/blob/5025fc1eda686d9daae6dc8a5b045a501cde12d3/MdePkg/Include/Library/PrintLib.h#L95

closed 2025-09-04 by ddaskala