tianocore/edk2 #11460: [Bug]: <title>OVMF secure boot settings unusable with serial console

CreeperKong, 2025-09-02:
### Is there an existing issue for this?

- [x] I have searched existing issues

### Bug Type

- [x] Firmware
- [ ] Tool
- [ ] Unit Test

### What packages are impacted?

OvmfPkg

### Current Behavior

There is no way to save secure boot settings when serial console is used.

### Expected Behavior

Give a way to save secure boot settings like "Commit changes and Exit".

### Steps To Reproduce

1. Start a QEMU instance with OVMF using serial console display
2. Press ESC to get into UiApp
3. Go to Device Manager and than Secure Boot
4. No way to save changes

### Build Environment

```markdown
INDEPENDENT
```

### Version Information

```text
f718b0f
```

### Urgency

Low

### Are you going to fix this?

Someone else needs to fix it

### Do you need maintainer feedback?

No maintainer feedback needed

lgao4, 2025-09-18:
@CreeperKong , which hot key doesn't work in serial console? Do you confirm Tty tool setting to make sure it match the terminal type for the platform setting.

CreeperKong, 2025-12-27:
Now fixed with capability to press F10 to save

closed 2025-12-27 by CreeperKong