tianocore/edk2 #11466: [Documentation]: <title> Need to update the "Usage" section of edk2/OvmfPkg/IntelTdx/README.md

liuw1, 2025-09-04:
### Request Description

1.	Refer to [docs/system/i386/tdx.rst · master · QEMU / Qemu · GitLab](https://gitlab.com/qemu/qemu/-/blob/master/docs/system/i386/tdx.rst?ref_type=heads): Since TDX doesn't support readonly memslot, TDVF cannot be mapped as pflash device and it actually works as RAM. "-bios" option is chosen to load TDVF.
2.	Update the KVM & QEMU URL to upstream URL.
https://git.kernel.org/pub/scm/linux/kernel/git/torvalds/linux.git 
https://gitlab.com/qemu-project/qemu.git

### Are you going to make the change?

Someone else needs to make the change

### Do you need maintainer feedback?

No maintainer feedback needed

lgao4, 2025-09-18:
@jyao1 , could you help check it?

liuw1, 2026-01-04:
The issue has been fixed with commit 2a8d98d0a5e8d662726b79e632658919cc8321f7