tianocore/edk2 #11538: [Bug]: LoadImage causes a store access fault on RISCV64

48cf, 2025-09-21:
### Is there an existing issue for this?

- [x] I have searched existing issues

### Bug Type

- [x] Firmware
- [ ] Tool
- [ ] Unit Test

### What packages are impacted?

OvmfPkg

### Which targets are impacted by this bug?

RELEASE, DEBUG

### Current Behavior

!!!! RISCV64 Exception Type - 000000000000000F(EXCEPT_RISCV_STORE_ACCESS_PAGE_FAULT) !!!!
     t0 = 0x00000000000001000	     t1 = 0x00000000000001260
     t2 = 0x00000000000000093	     t3 = 0x00000000000000093
     t4 = 0x00000000000000094	     t5 = 0x0000000009EEC8DE0
     t6 = 0x0000000009AE3A000	     s0 = 0x00000000083FFF568
     s1 = 0x0000000009EE7F6EE	     s2 = 0x0000000009EE98208
     s3 = 0x00000000000000078	     s4 = 0x0000000009EE97158
     s5 = 0x00000000000001000	     s6 = 0x00000000000000001
     s7 = 0x0000000009AE35000	     s8 = 0x0000000009EE8BF94
     s9 = 0x0000000009F4C2098	    s10 = 0x0000000009EE97158
    s11 = 0x00000000000000036	     a0 = 0x0AFAFAFAF00000000
     a1 = 0x00000000000000020	     a2 = 0x000000000000000AF
     a3 = 0x00000000083FFF000	     a4 = 0x0AFAFAFAFAFAFAFAF
     a5 = 0x0AFAFAFAFAFAFAFAF	     a6 = 0x00000000000000001
     a7 = 0x00000000000000001	   zero = 0x00000000000000000
     ra = 0x00000000083671C5C	     sp = 0x0AFAFAFAFAFAFAFAF
     gp = 0x00000000000000000	     tp = 0x0000000008004C000
   sepc = 0x00000000083671C82	sstatus = 0x08000000200006120
  stval = 0x0AFAFAFAFAFAFAFAF

### Expected Behavior

The image is loaded and executed correctly.

### Steps To Reproduce

Try chainloading another EFI executable using the Limine bootloader, this used to work but it seems like a regression in EDK2. It uses standard LoadImage functionality so any application using that should be able to be used to reproduce this issue.

### Build Environment

```markdown
- OS(s): Arch Linux
- Tool Chain(s): GCC 15.2.0
```

### Version Information

```text
ac20e4398a6022288bb6311ae4b1a9e3c24870c4
```

### Urgency

High

### Are you going to fix this?

Someone else needs to fix it

### Do you need maintainer feedback?

No maintainer feedback needed

### Anything else?

Tried to bisect and find out which commit introduced this, but I couldn't build many intermediate commits.

```
[iretq@raptor edk2]$ git -C edk2 bisect bad 
There are only 'skip'ped commits left to test.
The first bad commit could be any of:
dbe17c79e7775eb1757939c5ec2a1711f7b1d691
977b68aa1efd400b54009f30c1f5c3225e9ea691
be053713c0b4638496c718ca3fb6faf39568f611
07552c31ad8b596e2c7725d2a0a573e47c30d2dc
4d80dc68c614365971d627ca836b39a773c98f8c
003c8887147a50aad8aa29aec1c554ed432b202e
2ff92cf2aebe3a468a9daa9d3fbae2e0b702cfac
ac20e4398a6022288bb6311ae4b1a9e3c24870c4
We cannot bisect more!
```

vlsunil, 2025-11-25:
A few bug critical fixes have been merged very recently. Could you please try again with latest edk2 master branch? A DEBUG build would provide better insights.

Mintsuki, 2026-10-25:
@tphan-ventana @vlsunil @andreiw I tracked this down. It reproduces on stock firmware without Limine, and the cause is in `OvmfPkg/RiscVVirt/Library/PlatformSecLib`.

### Cause

In the default PEI-less boot, the 4 KiB `SEC_HANDOFF_DATA_RESERVE_SIZE` is subtracted from the temporary RAM size twice, so the HOBs end one page below the top of the stack DXE actually runs on:

1. `SecEntry.S` subtracts it from `PcdOvmfSecPeiTempRamSize`, sets `sp` to `TempRamBase` plus the result (`0x83fff000`), and passes that reduced size to `SecStartupPlatform()`.
2. `SecStartupPlatform()` subtracts it again (`TempRamSize -= SEC_HANDOFF_DATA_RESERVE_SIZE;`). `SecPEIlessStartup()` therefore describes the temporary RAM as `[0x82000000, 0x83ffe000)` in the PHIT and the stack as `[0x83ffa000, 0x83ffe000)` in the stack HOB.
3. `LoadDxeCoreFromFv (NULL, 0)` enters DXE on that same stack. DXE, BDS and every image they start run on a stack whose top page, `[0x83ffe000, 0x83fff000)`, no HOB describes, so the memory map reports it as `EfiConventionalMemory`.

`AllocatePages()` will then hand out the page holding the live stack. When that page is freed, `FreePages()` fills it with `0xAF`, because `RiscVVirtQemu.dsc` defaults to `DEBUG_ON_SERIAL_PORT = TRUE`, which sets `PcdDebugPropertyMask` to `0x2f` in RELEASE builds too. That is where the `sp = 0xAFAFAFAFAFAFAFAF` in the dumps above comes from. Limine runs into it because it allocates all free memory and gives it back with `FreePages()` before chainloading. Ordinary `AllocateAnyPages()` requests are served from the top of that free range, which is why other loaders rarely notice.

Both subtractions came in with 977b68aa1efd ("OvmfPkg/RiscVVirt: Add PlatformSecLib library"), part of the bisect range in the original report. Dropping one of them, for instance the one in `SecStartupPlatform()`, should make the HOBs cover the stack. I have not built and tested a patched firmware.

### Reproducer

A standalone application. Put it in `src/main.c` of https://github.com/PicoEFI/picoefi-c-template and run `make ARCH=riscv64 run`, which boots it with a RELEASE build of edk2-stable202608 that shows the bug too. Or boot the resulting `bin-riscv64/efi-template.efi` any other way.

<details><summary>src/main.c</summary>

```c
/*
 * Reproducer for tianocore/edk2#11538: on OvmfPkg/RiscVVirt the stack an
 * image is started on lies in memory the memory map reports as
 * EfiConventionalMemory, so AllocatePages() hands it out.
 *
 * Build with https://github.com/PicoEFI/picoefi-c-template: replace
 * src/main.c with this file, then "make ARCH=riscv64 run".
 */
#include <efi.h>

static EFI_SYSTEM_TABLE *st;

static void print(const char *s) {
    EFI_CHAR16 buf[128];
    EFI_UINTN n = 0;
    while (*s != 0 && n < sizeof(buf) / sizeof(buf[0]) - 2) {
        if (*s == '\n') {
            buf[n++] = '\r';
        }
        buf[n++] = (EFI_CHAR16)*s++;
    }
    buf[n] = 0;
    st->ConOut->OutputString(st->ConOut, buf);
}

static void print_hex(EFI_UINT64 v) {
    char buf[19] = "0x";
    int n = 2;
    int shift = 60;
    while (shift > 0 && ((v >> shift) & 0xf) == 0) {
        shift -= 4;
    }
    for (; shift >= 0; shift -= 4) {
        buf[n++] = "0123456789abcdef"[(v >> shift) & 0xf];
    }
    buf[n] = 0;
    print(buf);
}

EFI_STATUS efi_main(EFI_HANDLE image, EFI_SYSTEM_TABLE *system_table) {
    (void)image;
    st = system_table;
    EFI_BOOT_SERVICES *bs = st->BootServices;

    EFI_UINT64 frame = (EFI_UINT64)(EFI_UINTN)__builtin_frame_address(0);
    print("\nframe ");
    print_hex(frame);

    static EFI_UINT8 map[32768];
    EFI_UINTN size = sizeof(map), key, desc_size;
    EFI_UINT32 desc_version;
    if (bs->GetMemoryMap(&size, (EFI_MEMORY_DESCRIPTOR *)map, &key, &desc_size, &desc_version) == EFI_SUCCESS) {
        for (EFI_UINTN off = 0; off + desc_size <= size; off += desc_size) {
            EFI_MEMORY_DESCRIPTOR *d = (EFI_MEMORY_DESCRIPTOR *)(map + off);
            EFI_UINT64 end = d->PhysicalStart + d->NumberOfPages * EFI_PAGE_SIZE;
            if (frame >= d->PhysicalStart && frame < end) {
                print(" is in [");
                print_hex(d->PhysicalStart);
                print(", ");
                print_hex(end);
                print("), type ");
                print_hex(d->Type);
                print(d->Type == EfiConventionalMemory ? " (EfiConventionalMemory: BUG)\n" : "\n");
            }
        }
    }

    EFI_PHYSICAL_ADDRESS page = frame & ~(EFI_UINT64)(EFI_PAGE_SIZE - 1);
    EFI_STATUS status = bs->AllocatePages(EfiAllocateAddress, EfiLoaderData, 1, &page);
    print("AllocatePages(AllocateAddress) of its page: ");
    print_hex(status);
    if (status != EFI_SUCCESS) {
        print(" (refused, as it should be)\n");
    } else {
        print(" (granted: BUG)\n");
        print("Freeing it again, which firmware that clears freed memory fills with 0xAF...\n");
        bs->FreePages(page, 1);
        print("...still running: this firmware does not clear freed memory\n");
    }

    for (;;) {
    }
}
```

</details>

Output with RiscVVirtQemu built from current master (0a503bc2f5bd, RELEASE), QEMU 11.1.2, `-m 2G`:

```
frame 0x83ffeae0 is in [0x83ffe000, 0xff028000), type 0x7 (EfiConventionalMemory: BUG)
AllocatePages(AllocateAddress) of its page: 0x0 (granted: BUG)
Freeing it again, which firmware that clears freed memory fills with 0xAF...
!!!! RISCV64 Exception Type - 000000000000000C(EXCEPT_RISCV_INST_ACCESS_PAGE_FAULT) !!!!
...
    s11 = 0x000000000838A9710	     a0 = 0x00000000083FFE000
     a1 = 0x00000000000000020	     a2 = 0x000000000000000AF
     a3 = 0x00000000083FFF000	     a4 = 0x00000000000000000
     ra = 0x0AFAFAFAFAFAFAFAF	     sp = 0x0AFAFAFAFAFAFAFAF
...
```

For comparison, ArmVirtQemu from the same commit (OvmfPkgX64 behaves the same):

```
frame 0x47385a10 is in [0x46d55000, 0x47386000), type 0x4
AllocatePages(AllocateAddress) of its page: 0x800000000000000e (refused, as it should be)
```